particle-iot-archived/particle-dev #67: v0.0.20 does not display Spark.variable/Spark.function window on Windows 8.1 tablet

ScruffR, 2015-01-10:
I have tried different things to get this to work - deinstalled/reinstalled severalt times.
v0.0.19 did work.

suda, 2015-01-13:
@ScruffR I'm having a very hard time to replicate this behaviour. Could you try removing `~/.spark` and `~/.sparkdev` dirs? Also, try opening new window in Developer mode: View -> Developer -> Open In Dev Mode...

ScruffR, 2015-01-13:
@suda, tried it all but didn't help.

My system:
Win 8.1 Pro 64bit German
Lenovo Thinkpad Tablet 10
4GB RAM
128GB "HDD"
Now my earlier "@" and " \" problem is back - I have to find out how it was solved back then  ;-)

suda, 2015-01-13:
Could you try opening Developer Tools (View -> Developer -> Toggle Developer Tools) and enter in console:
```
atom.workspace.open('spark-dev://editor/cloud-functions').then(function(obj){console.log(obj)})
```
and pasting the result?

"@" problem was solved by removing `symbols-view` package but since last update Atom fails to work without it installed on Windows. You can go to Settings View and disable it.

ScruffR, 2015-01-13:
Ahh, that did the trick, this way it pops up

The result
```
atom.workspace.open('spark-dev://editor/cloud-functions').then(function(obj){console.log(obj)})
Object {promiseDispatch: function, valueOf: function, inspect: function, passByCopy: function, join: function…}inspect: function () {promiseDispatch: function (resolve, op, operands) {source: ObjectvalueOf: function () {__proto__: Promise
[div#spark-dev-cloud-functions-container, element: div#spark-dev-cloud-functions-container, functions: jQuery.fn.jQuery.init[1], emitter: Emitter, subscriber: Subscriber, lastOpened: 1421167384485…]0: div#spark-dev-cloud-functions-containerelement: div#spark-dev-cloud-functions-containeremitter: Emitterfunctions: jQuery.fn.jQuery.init[1]lastOpened: 1421167384485length: 1subscriber: Subscriber__proto__: ctor[0] VM782:2
```

suda, 2015-01-13:
@ScruffR you mean the panel showed up? Does it work now when invoked from menu? Does it still work after restarting Dev?

ScruffR, 2015-01-13:
Yes, as soon I put you command in, the panel popped up, but no - after a restart it still doesn't work.

suda, 2015-01-14:
@ScruffR Could you try to replace `Spark Dev\resources\app\node_nodules\spark-dev-cloud-functions\lib\spark-dev-cloud-functions.js` and `Spark Dev\resources\app\node_nodules\spark-dev-cloud-functions\lib\spark-dev-cloud-functions-view.js` (inside Spark Dev dir) with those two: http://cl.ly/1G1x2u1A3f1r open functions from menu and paste console contents?

ScruffR, 2015-01-14:
Tried it, didn't change a thing :-(

suda, 2015-01-14:
@ScruffR yeah, it only should output more messages to the console. Could you paste the results?

ScruffR, 2015-01-14:
I see, OK then here it is ;-)

```
atom.workspace.open('spark-dev://editor/cloud-functions').then(function(obj){console.log(obj)})
Open  spark-dev://editor/cloud-functions c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:17
Can be handled c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:19
CloudFunctionsView setup c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:49
listFunctions [] c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:125
Object {promiseDispatch: function, valueOf: function, inspect: function, passByCopy: function, join: function…}
spark.login c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:62
Object {accessToken: "...46a800eb..."} c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:63
[div#spark-dev-cloud-functions-container, element: div#spark-dev-cloud-functions-container, functions: jQuery.fn.jQuery.init[1], emitter: Emitter, subscriber: Subscriber, lastOpened: 1421247367350…] VM773:2
spark-dev:core-status-updated c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:77
listFunctions [] c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:125
```

suda, 2015-01-14:
@ScruffR could you restart the Dev, try to open using `Spark -> Show cloud functions` menu and paste results from such try?

ScruffR, 2015-01-14:
Sorry, didn't get that :blush: 
But now I've tried it and I don't get any result in the console, just as if the menu item wouldn't fire any event.

suda, 2015-01-14:
@ScruffR great! This narrows it out :) Could you show command palette (`ctrl+shift+p`) type "cloud view" and take a screenshot of resulting list? Also try selecting "Show Cloud Functions" and press Enter. Does this produces any console output?

ScruffR, 2015-01-14:
Yes the window pops up when selecting it there

[image]

suda, 2015-01-14:
@ScruffR ok, we're getting closer. Now:
1. Replace `spark-dev-cloud-functions.js` with http://cl.ly/code/3M1C2f421s1W 
2. Restart Dev
3. Open Console
4. Try to Show Cloud Functions from menu 
5. Paste console results here.

ScruffR, 2015-01-14:
Tried that. 
When I opened the console I got these new lines

```
Activate c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:9
Window load time: 12899ms VM60 index.js:39
spark-dev activated c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:12
sparkDevCloudFunctionsView created c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:17
```

But when trying to "Show Cloud Functions ..." nothing happened.
Only when I closed and reopened the console suddenly I had 155 errors

```
file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/buffered-process.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/dev-live-reload/lib/watcher.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/pane-container-view.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/panel-container-element.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/module-cache.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/bracket-matcher/lib/tag-finder.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/highlights-component.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/select-list-view.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/pane-axis.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/menu-manager.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/tabs/lib/tab-view.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/release-notes-spark/lib/release-notes-status-bar.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/scroll-view.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/pane.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/workspace-element.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/spark-dev/lib/utils/menu-manager.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/syntax.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/display-buffer.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/marker.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/highlight-component.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/git-diff/lib/git-diff-view.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/welcome-spark/lib/welcome.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/bookmarks/lib/bookmarks.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/text-editor-element.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/menu-helpers.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/tokenized-line.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/project.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/background-tips/lib/background-tips.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/exports/atom.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/decoration.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/spark-dev/lib/views/status-bar-view.js.map
[... cut ...]
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/snippets/lib/snippet.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/src/panel-element.js.map
Failed to load resource: net::ERR_FILE_NOT_FOUND file:///C:/Users/suda/AppData/Local/Temp/atom-build/Atom/resources/app/node_modules/fuzzy-finder/lib/main.js.map
```
This is reproducable and I don't understand why the program would look for a directory `C:/Users/suda` on my machine.

suda, 2015-01-14:
@ScruffR `.map` file errors aren't related to this problem (those are because you've replaced original `.js` files).

Ok, lets replace `spark-dev-cloud-functions.js` with http://cl.ly/code/3V1X2j0F2P0n for the last time and repeat all previous steps.

ScruffR, 2015-01-17:
Sorry for the late response. 
I've put in this new file and again the window/pane does not pop up.
And this was what the console gave me on opening it
```
Activate c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:9
Window load time: 3201ms VM60 index.js:39
spark-dev activated c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:12
sparkDevCloudFunctionsView created c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:17
spark-dev:update-menu c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:32
User is logged in c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:34
{"label":"Spark","submenu":[{"label":"Log out scruff.r@sbg.at","command":"spark-dev:logout"},{"type":"separator"},{"label":"Select device...","command":"spark-dev:select-device"},{"type":"separator"},{"label":"Claim device...","command":"spark-dev:claim-device"},{"label":"Identify device...","command":"spark-dev:identify-device"},{"label":"Setup device's WiFi...","command":"spark-dev:setup-wifi"},{"type":"separator"},{"label":"Compile in the cloud","command":"spark-dev:compile-cloud"},{"type":"separator"},{"label":"Show serial monitor","command":"spark-dev:show-serial-monitor"},{"label":"Show cloud functions","command":"spark-dev-cloud-functions-view:show-cloud-functions"}]} c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:41
spark-dev:update-menu c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:32
User is logged in c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:34
{"label":"Spark","submenu":[{"label":"Log out scruff.r@sbg.at","command":"spark-dev:logout"},{"type":"separator"},{"label":"Select device...","command":"spark-dev:select-device"},{"type":"separator"},{"label":"Claim device...","command":"spark-dev:claim-device"},{"label":"Identify device...","command":"spark-dev:identify-device"},{"label":"Setup device's WiFi...","command":"spark-dev:setup-wifi"},{"type":"separator"},{"label":"Compile in the cloud","command":"spark-dev:compile-cloud"},{"type":"separator"},{"label":"Show serial monitor","command":"spark-dev:show-serial-monitor"},{"label":"Show cloud functions","command":"spark-dev-cloud-functions-view:show-cloud-functions"}]} c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:41
spark-dev:update-menu c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:32
User is logged in c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:34
{"label":"Spark","submenu":[{"label":"Log out scruff.r@sbg.at","command":"spark-dev:logout"},{"type":"separator"},{"label":"Select device...","command":"spark-dev:select-device"},{"label":"Rename ScruffyWork...","command":"spark-dev:rename-device"},{"label":"Remove ScruffyWork...","command":"spark-dev:remove-device"},{"label":"Show cloud variables and functions","command":"spark-dev:show-cloud-variables-and-functions"},{"label":"Flash ScruffyWork via the cloud","command":"spark-dev:flash-cloud"},{"type":"separator"},{"label":"Claim device...","command":"spark-dev:claim-device"},{"label":"Identify device...","command":"spark-dev:identify-device"},{"label":"Setup device's WiFi...","command":"spark-dev:setup-wifi"},{"type":"separator"},{"label":"Compile in the cloud","command":"spark-dev:compile-cloud"},{"type":"separator"},{"label":"Show serial monitor","command":"spark-dev:show-serial-monitor"},{"label":"Show cloud functions","command":"spark-dev-cloud-functions-view:show-cloud-functions"}]} c:\Program Files (x86)\Spark Dev\resources\app\node_modules\spark-dev-cloud-functions\lib\spark-dev…:41
```
But when trying to open "Show cloud variables and functio..." nothing happened in the console either.

suda, 2015-01-18:
Hmmm where do you see "Show cloud variables and functions"? Because in 0.0.20 it was replaced with two different options: "Show cloud functions" and "Show cloud variables".

I'm afraid it's a Atom's issue with calling commands on menu. Would you mind using Command Palette instead menu until we fix it?

ScruffR, 2015-01-18:
I see :eyes: !

But I do still have the old menu item and the new ones are nowhere to be found.
This way it wouldn't be surprising that the two seperate commands are not called.

[image]

suda, 2015-01-18:
Wow! It looks like 0.0.19 mashed together with 0.0.20. Could you uninstall Dev, if any files are left in Program Files ZIP and send them to me? After this just remove whole `Spark Dev` directory from Program Files and install fresh copy.

ScruffR, 2015-01-18:
Actually it seems that unsinstalling Spark Dev doesn't remove any files.
When I do uninstall via Windows Control Panel the setup program immediatley returns.
If I should still send you the zip you need to prepared for a 61MB file.

suda, 2015-01-18:
Ok, it's installer's fault. You don't have to send it to me. Just remove whole dir and I'll fix the uninstaller in next version :)

ScruffR, 2015-01-18:
Now I have completely removed the "Spark Dev" folder in "Program Files" and also all the ".spark*" folders in my account directory, but I still have the old combined menu item.
I had downloaded the install exe again, just to be on the save side, but to no avail.

suda, 2015-01-19:
0.0.20 installer is here: https://www.spark.io/dev

Check if you deleted "Spark Dev" from both "Program Files" and "Program Files (x86)".

ScruffR, 2015-01-19:
Still no fun :anguished:

ScruffR, 2015-04-28:
Just tried to download 0.0.21, which should fix my issue, but the link in Spark Dev seems broken and I can't find it on github either.

suda, 2015-04-28:
@ScruffR 0.0.21 isn't released yet. Should be available tomorrow at the latest.

suda, 2015-05-02:
This should be now fixed in [0.0.21](https://github.com/spark/spark-dev/releases/tag/v0.0.21)

closed 2015-05-02 by suda