particle-iot-archived/particle-dev #70: Firmware Upload RGB led problem when used RGB.control(true);

hallard, 2015-01-24:
Hi there,
Not really an issue, just to let you know that when user take control of RGB LED with  
`RGB.control(true);`  
Then when we upload new firmware, during the 1st phase of firmware transfer (the upload) take control of the LED just for blinking (ie it blink only the led)
So if the led was off at thiis precise time we don't see anything, else the LED start blinking with the color it had just before upload started instead of magenta.

m-mcgowan, 2015-01-24:
@hallard  Thanks for the report. This is a known issue with the firmware, and is being tracked here - https://github.com/spark/firmware/issues/362.

suda, 2015-01-25:
I think you should

closed 2015-01-25 by suda