particle-iot-archived/particle-dev #71: Library Folder

hallard, 2015-02-18:
Hi there,
I'm using lot of libraries with Spark dev, and using this structure 
```
MyProject/project.ino
MyProject/mylib/lib.h
MyProject/mylib/lib.cpp
```
Everything is working fine and I'm dealing with this for long time. But, unfortunately I'm trying to port a Arduino Library with lot of files and each cpp library file include the header like this
`#include <header.h>`
which make sense because on Arduino libraries are seen in a "system" folder but on Spark, this mean that compilation does not works until I replace the <> by "" like as follow
`#include "header.h"`
I can do it but this means I need to change each file each time there is a library update by arduino community.
Would it be possible to have (depending on the simple to implement)  

1) define in project folder a std libraries (whatever name in fact) folder such as
```
MyProject/project.ino
MyProject/libraries/mylib/lib.h
MyProject/libraries/mylib/lib.cpp
MyProject/libraries/otherlib/lib.h
MyProject/libraries/otherlib/lib.cpp
```
and tell the compiler to include the "./libraries" folder for compilation so like that it will find files included by `#include <header.h>` and not only by `#include "header.h"`

2) Or (my favorite) define in spark dev a local library folder and tell the compiler flags to include this local libraries folder for compilation ? This would solve all our libraries issues.

Hope this request makes sense

 Charles

suda, 2015-02-18:
Hi Charles,

this issue has deep roots in compile server and [library structure](https://github.com/spark/uber-library-example/issues/3). There was also discussion about it in the [community](https://community.spark.io/t/library-import-flattens-directory-structure-workaround/9915). It's something we want to fix really soon.

vshymanskyy, 2016-01-08:
Hi! This also affects Blynk library development, so I hope there is some progress already...

wmonroe, 2016-03-10:
Is there any update on putting libs in a central folder?

closed 2016-08-26 by suda

stratosmacker, 2016-11-03:
This bug seems to be open again....

After having nested directories work on the CLI for the last few months, today I can confirm that this bug is once again present.

suda, 2016-11-03:
@stratosmacker could you provide a code that doesn't compile properly in latest version of Dev?

commit 50791126 (2016-08-26): Support projects with nested directories

Close #71