particle-iot-archived/particle-dev #73: App code was invalid

crimsonclyde, 2015-02-24:
Same issue #63, #44 but Version 0.0.20 did not fix the problem.

I have a .git folder and a README.md file in it.
Every time I tried to build it fails with "App code was invalid" and "There where no compiler errors".

I had to move the .git folder and the README.md file out of the build folder where my .ino, .cpps and .h files are. Then it compiles and worked correct.

So, my question :) is the fix from #63 already patched into 0.0.20?

Cheers and keep up your good work @suda 
clyde

suda, 2015-02-24:
I'm very sorry @crimsonclyde but the fix hasn't been deployed yet. I'll do my best to release it this sprint.

crimsonclyde, 2015-02-24:
@suda : Take your time
Now I know that I´am not the problem :D

closed 2015-02-24 by crimsonclyde