particle-iot-archived/particle-dev #77: Compiler error messages should not be cropped

opyh, 2015-03-22:
Compiler error messages are cropped in the UI:

[image]

Have I overlooked a way to see the full error messages somewhere? That would be helpful when debugging, as the important bits are often at the (cut-off) ends of the messages.

Maybe they could just be logged to the console?

Even better would be having inline error messages, like in the [linter plugin](https://atom.io/packages/linter) ;)

rsteckler, 2015-04-22:
Have you found a way to work around this?  I'm having a hard time determinging what the compiler is complaining about.

suda, 2015-04-22:
This will be probably done by implementing #79 or some logging panel. Currently the main goal is updating to latest Atom and fully support upcoming Photon.

cdeadlock, 2015-09-06:
I had to do this pain in the arse to see full errors for another problem #110 

In atom go to the menu -> View -> Developer -> toggle developer tools
Its just like chrome or chromium developer tools: you can click on the magnifying glass and click on a UI element, and see the source for the full information

newzealandpaul, 2015-10-09:
This is a real pain. Is this being worked on?

suda, 2015-10-09:
Yes, we're going to introduce error console similar to one in Build IDE.

ponuganti, 2015-10-15:
This is rea big pain.. I get the error "build didn't produce binary Error: Command faild:.." I could see the full error .. this is minimum is ide that it shows what the error is ..

dhhagan, 2016-11-29:
Has this been implemented yet? This is the least lovable Particle Dev feature right now..

suda, 2016-11-29:
Unfortunately not yet. But it is on top of your Dev priorities!