particle-iot-archived/particle-dev #81: Console input not displayed on Win Spark Dev 0.0.21

ScruffR, 2015-05-09:
When I try to flash my firmware by use of the onboard console pane, I do see the gray "input your command here" message, but once I start entering text the input box goes blank and I can't see what I type.
But when I press <keyb>ENTER</keyb> the correct command is executed.

suda, 2015-05-15:
@ScruffR could you try the latest 0.0.23?

ScruffR, 2015-05-18:
That's a funny way to solve the problem ;-)
In 0.0.23 the console "button" seems to be gone.

suda, 2015-05-18:
@ScruffR could you share a screenshot?

ScruffR, 2015-05-18:
That's how it looks on my machine
[image]

suda, 2015-05-19:
You see these errors when you click `Flash` in upper left corner?

ScruffR, 2015-05-19:
No, these appear when I open the the Dev Tools pane, close it and open it again.
But these errors don't seem to have anything to do with the "missing" console button ;-)

suda, 2015-05-19:
Do you mean this one: https://www.dropbox.com/s/neiim6a5jntfhx6/Screenshot%202015-05-19%2012.07.34.png?dl=0 ? Because it's part of [tool-bar-main](https://atom.io/packages/tool-bar-main) which isn't bundled with Dev.

ScruffR, 2015-05-19:
Yes, that's the one, but up to v0.0.21 it appeard on the bottom status bar of my Windows "Spark Dev" without any furter ado ;-)
How can I add it now to 0.0.23?

suda, 2015-05-19:
Currently installing 3rd party packages is blocked by #80 but is to be fixed this sprint.

ScruffR, 2015-05-19:
OK, then I'll wait for that fix - thanks!

commit 0fdbb7b8 (2015-06-09): Updated packages and fixed #81
closed 2015-06-09 by suda