particle-iot-archived/particle-dev #83: Uncaught ReferenceError: logTile is not defined

cherltte, 2015-05-11:
Hi,

There are errors using spark dev.
Whenever I tried to flash code, I am getting two error messages

1.Uncaught ReferenceError: logTile is not defined
2.The error was thrown from the spark-dev package. spark-dev is out of date: 0.0.21 installed; 0.0.22 latest. Upgrading to the latest version may fix this issue.

And once it has been flashed (somehow it disregards errors and flash to core and core seems to work), but I don't get access to the core anymore. Meaning that the core is still breathing cyan led, but Spark Dev shows not connected. 

please help me out. this is very frustrating. 

and the detail message for the first error is as follow:
/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/lib/views/status-bar-view.js:193
Hide Stack Trace
ReferenceError: logTile is not defined
    at HTMLSpanElement.<anonymous> (/private/var/folders/99/3mvrfpwx19nb8w3ktwh3tfv00000gn/T/spark-dev11546-22723-1p6ryhs/node_modules/spark-dev/lib/views/status-bar-view.coffee:193:6)
    at HTMLSpanElement.opt.complete (/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:6693:12)
    at fire (/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:3099:30)
    at Object.self.fireWith [as resolveWith] (/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:3211:7)
    at Animation.tick (/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:6569:14)
    at jQuery.fx.tick (/Applications/Spark Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:6858:9)

suda, 2015-05-11:
Hi @cherltte, the error with `logTile` has been fixed and should be released this week.
As for core not showing up as online, it can be caused by code which blocks cloud functionalities (like endless `while` loop). Try [factory reset](http://docs.spark.io/connect/#appendix-factory-reset) and check your code.

closed 2015-05-15 by suda

jschrempp, 2015-05-18:
It says this is fixed six days ago. I just experienced the problem.

suda, 2015-05-18:
@jschrempp Have you updated to the latest version of Particle Dev? Try removing old Spark Dev and download latest version [from here](https://www.particle.io/dev).

jschrempp, 2015-05-29:
I wasn't aware that a new download of the dev app was needed. I have done that and the error has gone away. Thanks.