particle-iot-archived/particle-dev #86: Option to select between Particle.io or local cloud to send OTA updates to devices

chuank, 2015-05-23:
I'm curious if this can be implemented within Particle Dev.

Currently, a typical workflow for sending firmware OTAs to devices on a local cloud is:
1) write code on Particle Build Web IDE
2) compile on Web IDE, and download firmware.bin to local disk system
3) run `particle flash` to send firmware binary to desired device(s)

It'll be great if:
0) user has the option of defining a local cloud url to send OTA updates to
1) code is written in Particle Dev
2) `CMD-U` gets the firmware compiled via the Particle.io cloud _and_ then sent as an OTA update over a local spark-server

Please note that I'm _not_ referring to compiling locally; the benefits of a Particle.io remote compile (easily accessible community libraries, for a start) still far outweighs the task of say non-technical users trying to set up a local compiler.

This should enhance productivity immensely as local cloud users currently have to switch between say Dev and cli and keep track of the .bin files being generated.

suda, 2015-05-26:
@chuank well there is an undocumented way of doing this by switching CLI config to local one: https://community.particle.io/t/tutorial-local-cloud-1st-time-instructions-20-march-15/5589

We plan to integrate config selection in Dev as a part of UI.

chuank, 2015-05-27:
Hi @suda, thank you for the links and update.

I am already using particle-cli's `apiUrl` configuration to switch to a local one, and it's how I perform cli-based firmware OTAs to my local Cores on my local cloud (I use the Build Web IDE on Particle.io to write the code with this workflow).

What I am referring to is the option to do *all* of the following *within* Dev:
 * write code (already possible)
 * verify/compile code via Particle.io compiler (ditto)
 * view/manage status of my devices on my local cloud
 * query variables/functions on local cloud
 * upload the newly-downloaded .bin to my local Cores via OTA

Really hope to see a UI-based cloud configuration selection within Dev, because it will provide users with the customisability of a local cloud, and still rely on the power and ease of OTA updates that the web IDE has. I feel that this is a key differentiator in offering up Dev as an alternative to the simpler web IDE.

If this is in the plan, perhaps allowing the user to also select compilers?

I'm suggesting all of this because while the current cli option to upload firmware works, it can be streamlined. A more complete integration goes a long way towards improving workflow efficiency, and Dev seems to be the right tool to integrate this workflow.

suda, 2015-05-27:
The idea is, that in the status bar, next to selected device, there would be a config selection. All cloud commands (compile/flash/query variables/functions) would be using currently selected config. Using Particle Cloud to compile and then flash the binary via local cloud would be possible, but with constant profile switching. What could be better for your scenario is #27 and using local config. This way, you'll be iterating much faster.

chuank, 2015-05-27:
agreed, makes a lot of sense to have that local compiler when the config selection is available. thanks for the tip!

suda, 2017-02-14:
We have a [particle-dev-profiles](https://github.com/spark/particle-dev-profiles) package now that provides authenticated API client and can be used for this purpose. I'm closing this in favor of [Add support for cloud profiles #5](https://github.com/spark/particle-dev-profiles/issues/5)

closed 2017-02-14 by suda