particle-iot-archived/particle-dev #89: Uploading after using the serial monitor breaks serial

balbano, 2015-06-05:
Reproducing the bug
---------------------------

```
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello World!");
  delay(1000);
}
```

On Windows 8.1, using Particle Dev and a Particle Photon, if I 

1. Compile and upload the above code in Particle Dev.
2. Open the Particle Dev serial monitor, click "Connect".
3. Close the serial monitor with the little "x" button
4. Make a change to the code.
5. Compile and upload again.
6. Try to reopen the serial monitor, the serial monitor can't connect to the Photon. (Clicking the connect button does nothing).

I can get the same problem if I open a serial connection through another application, such as PuTTy, then try to compile and upload from Spark Dev while that serial connection is still open. If I close PuTTy before uploading, then reopen after uploading, no problems.

A workaround
-----------------

In particle dev, this sequence has no problems:

1. Compile and upload in Particle Dev.
2. Open the Particle Dev serial monitor, click "Connect."
3. Disconnect by clicking the "Disconnect" button.
3. Make a change to the code.
4. Compile and upload again.
5. Reopen the serial monitor, the serial monitor connects just fine.

It seems like there are two issues
-----------------------------------------

1. Uploading to the Photon while a serial connection is open messes up the serial connection.
2. Closing the serial monitor panel in Particle Dev does not fully close the serial connection in the same way that clicking the "disconnect" button does.

balbano, 2015-06-05:
I've never done any hacking around in atom, so I may be way off here, but could the serial monitor panel use [onDidDestroy](https://atom.io/docs/api/v0.206.0/Panel#instance-onDidDestroy) to call the disconnect() function when you close the window so that it behaves the same as the disconnect button?

suda, 2015-06-05:
Thanks for this detailed report Brendan! It seems that this could be solved by disconnecting before flashing the device. Problem is we don't know which device is connected via USB and is it the same device you're trying to flash. We could do it anyway and add option in settings to disable this behaviour. Would that be ok?

Disconnecting on closing serial monitor panel should be implemented. Added to backlog.

balbano, 2015-06-05:
Having an option to disconnect any connected devices when you flash would be great!

balbano, 2015-09-15:
I just tested this out today and it looks like it's been fixed in some version of Particle Dev between when I made the issue and now! Thanks guys!

closed 2015-09-15 by balbano