particle-iot-archived/particle-dev #90: Migrating project from core to photon issues

hallard, 2015-06-09:
Hi there, 

I plugged my new Photon (really nice boards ;-) )on my shield board in place of old Spark Core device, then I tried to compile my project with particle dev 0.0.23. Here are the issues I've got (I just changed from a Core device to a Photon device in the IDE)

* on I2C wire, the defined CLOCK_SPEED_100KHZ and CLOCK_SPEED_400KHZ seems to be undefined throwing error CLOCK_SPEED_400KHZ was not declared in this scope If I call Wire.setSpeed(CLOCK_SPEED_400KHZ);

* I was unable to compile my project because I got another error, undefined reference to "Spark_Idle_Events", I wasn't calling this anywhere but found it was on some of my SPARK_WLAN_Loop(), so I replaced theese by Spark.process(), errors are gone. It could be interesting to depreciate SPARK_WLAN_Loop() for Photon in documentation if it's not already done

* this one could not be new but it would be useful to know on which file the error is when you have, because searching in all sources files can be a pain

That's it, got 2 I2C devices, one RFM69, 2nd serial, everything seems to work fine, nice job guys ;-)

m-mcgowan, 2015-06-09:
Hey @hallard! Thanks for these reports. Fortunately they are known issues and will be fixed in the next release later this week!

As  a workaround to the I2C click speed issue, please add `#include "i2c_hal.h" to your sketch, and the code will compile.

Cheers,
mat.

hallard, 2015-06-09:
Thanks Mat, good to know.

For information adding including "i2c_hal.h" or <i2c_hal.h> did not worked, compilation tell me no such file or directory i2c_hal.h (I'm on spark dev)
Cheers,
Charles

m-mcgowan, 2015-06-09:
strange, the compilation is done in the cloud, against the current Photon firmware. 

Here's a link to the problem and the solution.

https://community.particle.io/t/compilation-of-core-code-for-photon-gives-errors/12195/10?u=mdma

hallard, 2015-06-09:
thanks Mat,
Sorry about that, in the meantime I changed platform to core that was why it wasn't working, switched back to photon, now it found the file i2c_hal.h ;-)
this bring me a question, how can I know if I'm compiling for core or photon ? I'm pretty sure there is a defined symbol such as SPARK or PHOTON ?

m-mcgowan, 2015-06-09:
Please see the thread I linked to, it discusses that.

hallard, 2015-06-09:
Mat,
well, I was wondering where to do my initial post issue, should have done it on the forum instead of github, good to know for next time ;-)
thanks again

m-mcgowan, 2015-06-09:
Yep, for sure, the forum is a good place to test the waters - and then if it's not a known issue then we go ahead and create a github issue for it.

Glad you're up and running, will close this issue now! ;-)

closed 2015-06-09 by m-mcgowan