particle-iot-archived/particle-dev #95: Web IDE/Particle Build seems to treat libraries differently between Core and Photon

ScruffR, 2015-07-02:
http://community.particle.io/t/how-to-include-a-library-because-im-failing/12931/8

In this thread @mumblepins found that a lib he ported did build against the Photon but not against Core due to using `#include <....>` instead of `#include "...."`.
The behaviour should be consistent accross platforms.

zsup, 2015-07-02:
This sounds like a firmware issue rather than an IDE issue; @ScruffR can I ask you to open an issue in `spark/firmware` instead?

technobly, 2015-07-02:
Pretty sure this would be a compile-service issue with the preprocessor.  This exact issue has come up in the past.  cc: @dmiddlecamp ;-)

dmiddlecamp, 2015-07-02:
the only difference between these for the build farm is essentially how the makefile treats the library files when they're included in the build, so it's possible something reverted in the newer makefile

zsup, 2015-07-02:
so if it's a makefile issue then it's a firmware issue, yes?

dmiddlecamp, 2015-07-02:
I _think_ so, but I could be wrong.  Either way some unit testing on the build farm repo that confirms the desired behavior would be cool, then we can be confident we don't regress, even if we make the change in the firmware repo.

ScruffR, 2015-07-02:
So if it could be either a FW or an IDE issue (and I'm not in the position to tell), can I ask you Dave to open the FW issue if it turns out to be FW rather than me doing it now and then it turns out it wasn't after all ;-)

dmiddlecamp, 2015-07-02:
looking back at this, I don't think we supported the `<...>` syntax before, so it sounds like the newer makefile is more forgiving instead of less forgiving, so I think this isn't a regression.  Does that sound right?

technobly, 2015-07-02:
We used to include <...> some files in the way past, it was actually a thing... and at some point it started failing.  Search won't find things like this, but I looked through some of my old libraries.  Check it out... I know this compiled once upon a time:
https://github.com/technobly/SparkCore-RF24/blob/master/RF24.cpp#L22

suda, 2017-02-14:
With the arrival of libraries V2, this has been now mitigated.

closed 2017-02-14 by suda