particle-iot-archived/particle-dev #97: Compiler error messages reference line number but not file name

osmithy, 2015-07-10:
When I compile and get an error, it shows the line number but not the file name. For example, I just tried to compile and got the following error:

'PIN_MAP' was not declared in this scope
.:93:3

When I click on the error, it opens a blank document. Frustrating!!

jlkalberer, 2015-07-12:
Yeah, I am also seeing this.  It's very hard to develop anything with this going on.

suda, 2015-07-12:
I'm sorry about that. We'll fix it asap

jlkalberer, 2015-07-12:
Thanks a ton man.

jenesaisdiq, 2015-07-12:
Hey @osmithy are you using the SparkButton library with a Photon? That's the most common place I've seen PIN_MAP come up, and you can fix it by removing the library (click the Code icon in the sidebar and the X beside SparkButton under Included Libraries) and re-add it. That'll get you up to the current version where I've fixed the compatibility.

technobly, 2015-07-12:
Same deal for the NeoPixel library ;-)

osmithy, 2015-07-13:
I'm using a modified version of the Adafruit_SSD1351 library that I found here: https://github.com/pkourany/Adafruit_SSD1351_Library

One of the software people at my office helped fix it for me by implementing the fix [here](http://community.particle.io/t/photon-and-the-pin-map-challenge/12223), but I can only get it to compile with particle-cli, not the web IDE or Particle Dev. The errors I get are that pinHI and pinLO aren't declared in this scope, but they're definitely #defined earlier in the file. Oddness.

If you have time, it would be really awesome if you could fix this library and publish it so that I can use the web IDE again. I really like using it :)

paulmand3l, 2015-07-22:
Any updates on this? Still getting this error.

suda, 2015-07-23:
Hey @paulmand3l, sorry for delay. We'll release 0.0.25 with a fix for it this sprint.

commit 71e8b053 (2015-08-17): Fix #97 / IDE-94
closed 2015-08-17 by suda