particle-iot-archived/particle-dev #99: type_traits: No such file or directory

Techjar, 2015-07-20:
I get this error whenever I try to compile anything, and I'm not sure if this is a legitimate issue or if I've just missed something non-obvious. It gives the mostly meaningless line number ".:31:23". My code is just empty setup() and loop() methods, and does have the "application.h" include.

kennethlimcp, 2015-07-20:
i believe only `.ino`, `.h`, `,cpp` files are allowed

suda, 2015-07-20:
@Techjar could you zip the project dir and share it?

Techjar, 2015-07-20:
There's nothing particularly unique about it, but sure.
http://owncloud.techjargaming.com/public.php?service=files&t=FjP4SKdvlZBxrJC

kennethlimcp, 2015-07-20:
IMO `.c` is not supported

suda, 2015-07-20:
@Techjar when I renamed `main.c` to `application.cpp` it compiles. Sorry the default filename isn't specified in docs. Will add it right now :)

Techjar, 2015-07-20:
Does it need to be `application.cpp` or will `main.cpp` work as well?

suda, 2015-07-20:
Any `.cpp` should work. Naming your code `.c` suggests the compiler your code is in C but the [firmware](https://github.com/spark/firmware) is written in C++ and the `application.h` header uses a lot of its features (like `extern "C"`). This is why you can't mix those.

Techjar, 2015-07-20:
Ah. Weird, because `.c` used to work in older versions.

closed 2015-07-23 by suda