particle-iot-archived/particle-dev #101: Atom is out of date but update says "All of your installed Packages are up to date!"

niccokunzmann, 2015-07-25:
[Enter steps to reproduce below:]

1. Connect to your Photon.
2. Write the program `test.cpp`
    ```
void setup () {

}

void loop() {

}
```
3. Paticle -> Compile in the Cloud

**Atom Version**: 0.0.24
**System**: Microsoft Windows 7 Professional 
**Thrown From**: [spark-dev](https://github.com/spark/spark-dev) package, v0.0.23

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, uv_chdir

```
At C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\babel-core\node_modules\regenerator\node_modules\commoner\node_modules\graceful-fs\polyfills.js:14

Error: ENOENT: no such file or directory, uv_chdir
  at Error (native)
  at process.chdir (C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\babel-core\node_modules\regenerator\node_modules\commoner\node_modules\graceful-fs\polyfills.js:14:9)
  at C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:661:21
  at Object.module.exports.projectRequired (C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:369:14)
  at C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:639:24
  at Object.module.exports.loginRequired (C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:353:14)
  at Object.module.exports.compileCloud (C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:637:19)
  at atom-workspace.disposables.add.atom.commands.add.spark-dev:compile-cloud (C:\Program Files (x86)\Particle Dev\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:132:26)
  at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Program Files (x86)\Particle Dev\resources\app.asar\src\command-registry.js:238:29)
  at CommandRegistry.handleCommandEvent (C:\Program Files (x86)\Particle Dev\resources\app.asar\src\command-registry.js:3:61)
  at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Program Files (x86)\Particle Dev\resources\app.asar\src\command-registry.js:153:19)
  at EventEmitter.<anonymous> (C:\Program Files (x86)\Particle Dev\resources\app.asar\src\window-event-handler.js:70:30)
  at emitOne (events.js:77:13)
  at EventEmitter.emit (events.js:166:7)

```

### Commands

```
     -1:21.6.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
     -1:21.5.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always)
     -1:21.5.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
     -1:19.8.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -1:19.8.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -1:19.8.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -1:19.8.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:50.9.0 spark-dev:compile-cloud (div.settings-view.pane-item)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```
# Aditional Information

- Spark Photon, device is claimed
- When creating the issue automatically I get 
`The Error wwas thrown fro mthe PSark dev package . Atom is outof date: 0.0.24 installed; 1.02 latest. Upgrading to the latest version may fix the issue.`

- It was a freshly installed particle dev environment.

closed 2015-08-18 by suda

niccokunzmann, 2015-08-18:
What is the problem?

suda, 2015-08-18:
It has been resolved with [v0.0.25](https://github.com/spark/spark-dev/releases/tag/v0.0.25) :)