particle-iot-archived/particle-dev #102: Can't Find a Library File (Sometimes) During Verification

craigfoo, 2015-07-28:
I have my main sketch in the root directory as main.ino. I then have a lib folder with two sub folders which have some .h and .cpp files. I've already had success with verifying and sending to the Photon but it has since flaked out and says that it couldn't find one of my .h files. It's clearly included in main.ino and it's clearly in the file structure. It worked before! Help?

suda, 2015-07-28:
@craigfoo what is the path to the lib in `#include` statement? Because even if your file is in `foo/bar.h` you have to include it as `#include "bar.h"`

craigfoo, 2015-07-28:
Ha! Ok, that worked but it was strange that it worked for awhile, then didn't. All my includes are now
'#include "bar.h"'

closed 2015-09-18 by suda