particle-iot-archived/particle-dev #110: Compiler issue

aphanson, 2015-09-01:
I'm seeing an error "Compiler timed out or encountered an error" and when I click on it it says "There were no compile errors". So I take it the compiler is timing out. Any help would be great. Thanks

dmiddlecamp, 2015-09-01:
HI @aphanson ,

  It's possible there is a special character or something in your filename that is preventing your app from being compiled.  Can you provide some more context?  Are you compiling for a photon or core?  Are you using the latest version of Dev?  Are there any special characters in the filenames?

Thanks,
David

cdeadlock, 2015-09-06:
I just had a similar situation: 

my tool-bar says
```
build didn't produce binary Error: Command failed: /spark/co...
```

clicking on it shows the compile-errors view which has
```
there were no compile errors
```

I had to open developer tools and view the source to find out the actual error, which isn't being parsed properly or something
```
build didn't produce binary Error: Command failed: /spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/dispatcher.cpp: In member function 'void Dispatcher::setLog(Logger*)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/dispatcher.cpp:21:12: warning: deleting object of polymorphic class type 'Logger' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete log;
            ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/funcbase.cpp: In member function 'int Funcbase::matchfuncname(const char**, int*)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/funcbase.cpp:15:32: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (*len &lt; funcname.length() || !s.startsWith(funcname)){
                                ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/funcfreemem.cpp: In member function 'virtual int Funcfreemem::tryhandle(String)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/funcfreemem.cpp:38:33: warning: narrowing conversion of 'a.String::length()' from 'unsigned int' to 'int' inside { } [-Wnarrowing]
     int size[count] = {a.length()};
                                 ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/getclaim.cpp: In member function 'virtual int Getclaim::tryhandle(String)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/getclaim.cpp:66:48: warning: narrowing conversion of 'o.String::length()' from 'unsigned int' to 'int' inside { } [-Wnarrowing]
         int sizes[3] = {csize, tsize, o.length()};
                                                ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/logbase.cpp: In member function 'void Logbase::setLog(Logger*)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/logbase.cpp:7:12: warning: deleting object of polymorphic class type 'Logger' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete log;
            ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/resultcoordinator.cpp: In member function 'int ResultCoordinator::add(char**, int*, int)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/resultcoordinator.cpp:104:22: warning: 'u' may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (sizes[x] &gt; u-&gt;maxsize){
                      ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/savedata.cpp: In function 'void base64_encode(const unsigned char*, unsigned int, char*, int)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/savedata.cpp:47:17: warning: 'char_array_4[0]' is used uninitialized in this function [-Wuninitialized]
   unsigned char char_array_4[4];
                 ^
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/savedata.cpp:47:17: warning: 'char_array_4[1]' is used uninitialized in this function [-Wuninitialized]
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/savedata.cpp:47:17: warning: 'char_array_4[2]' is used uninitialized in this function [-Wuninitialized]
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/savedata.cpp:47:17: warning: 'char_array_4[3]' is used uninitialized in this function [-Wuninitialized]
../../../build/target/user/platform-6-m/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2//libuser.a(dispatcher.o): In function `Dispatcher::setLog(Logger*)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/dispatcher.cpp:27: undefined reference to `vtable for Logger'
../../../build/target/user/platform-6-m/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2//libuser.a(logbase.o): In function `Logbase::setLog(Logger*)':
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/logbase.cpp:8: undefined reference to `vtable for Logger'
[... cut ...]
/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/logger.cpp:8: undefined reference to `vtable for Logger'
collect2: error: ld returned 1 exit status
make: *** [/spark/compile_service/shared/workspace/6_photon_17_3/firmware/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2/a940dc7fb615eb3fd777aba10a736537e1ba5bc2dd67aab403260a4997d2.elf] Error 1
```

Atom says no update is required for the particle and spark-dev packages. I tried npm update in the spark-dev directory which may or may not have updated a couple of libraries... didn't seem to fix anything.

cdeadlock, 2015-09-06:
Also I think we should be able to see warnings somewhere without looking at the source even on success

suda, 2016-03-10:
Note to self: this seem to be caused by error during linking not compiling. Dev doesn't catch it thus shows whole output in status bar.

closed 2017-04-05 by suda