particle-iot-archived/particle-dev #115: First try to build other users project threw this

ScruffR, 2015-09-24:
I did a fresh install with the most recent Windows ParticleDevSetup.exe

```
[Enter steps to reproduce below:]

1. Fresh install
2. sign in
3. select Phiton
4. hit build

**Atom Version**: 1.0.15
**System**: ANDYTPT10
**Thrown From**: [spark-dev](https://github.com/spark/spark-dev) package, v0.0.26

### Stack Trace

Uncaught TypeError: Cannot read property '0' of undefined

```
At C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\decorators\unhandledRejection.js:80

TypeError: Cannot read property '0' of undefined
    at C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:761:58
    at tryCatchReject (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\makePromise.js:845:30)
    at runContinuation1 (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\makePromise.js:804:4)
    at Fulfilled.when (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\makePromise.js:592:4)
    at Pending.run (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\makePromise.js:483:13)
    at Scheduler._drain (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\Scheduler.js:62:19)
    at Scheduler.drain (C:\Users\Andy\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\node_modules\when\lib\Scheduler.js:27:9)
    at doNTCallback0 (node.js:416:9)
    at process._tickCallback (node.js:345:13)
```

### Commands

```
     -0:22.1.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
     -0:21.2.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.2.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:12.9.0 spark-dev:compile-cloud (atom-text-editor.editor.is-focused)
     -0:11.7.0 spark-dev:update-compile-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```
```

suda, 2015-09-24:
It's a duplicate of #105. Try opening directory instead of single file.

closed 2015-09-24 by suda

ScruffR, 2015-09-24:
I had opened a directory and just selected one of the files in there to edit it.