particle-iot-archived/particle-dev #117: Uncaught Error when compiling/uploading

ineil, 2015-09-30:
This error always appears when attempting to compile/upload on Mac OSX 10.10.5.
[image]

Note that I use Atom separately at its most recent version. I also noticed that in the Atom release history a similar sounding ENOENT issue is described as being resolved on Windows for version 1.0.17.

suda, 2015-10-05:
@ineil this is a generic version when trying to access non existent dir. Could you try opening just one directory (I see you have opened `config` and `particle`)?

suda, 2015-10-08:
This should be fixed in [v0.0.28](https://github.com/spark/spark-dev/releases/tag/v0.0.28).

commit 738e0192 (2017-01-24): Fix #117
closed 2015-10-08 by suda