particle-iot-archived/particle-dev #118: Error during atom update

cdeadlock, 2015-10-02:
```
Updating to “spark-dev@0.0.27” failed.Hide output…

> runas@2.0.0 install /tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher/node_modules/runas
> node-gyp rebuild

make: Entering directory `/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher/node_modules/runas/build'
  CXX(target) Release/obj.target/runas/src/main.o
  CXX(target) Release/obj.target/runas/src/runas_posix.o
  CXX(target) Release/obj.target/runas/src/fork.o
  SOLINK_MODULE(target) Release/obj.target/runas.node
  COPY Release/runas.node
make: Leaving directory `/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher/node_modules/runas/build'

> pathwatcher@3.3.3 install /tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher
> node-gyp rebuild

make: Entering directory `/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher/build'
  CXX(target) Release/obj.target/pathwatcher/src/main.o
  CXX(target) Release/obj.target/pathwatcher/src/common.o
  CXX(target) Release/obj.target/pathwatcher/src/handle_map.o
  CXX(target) Release/obj.target/pathwatcher/src/pathwatcher_linux.o
  SOLINK_MODULE(target) Release/obj.target/pathwatcher.node
  COPY Release/pathwatcher.node
make: Leaving directory `/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/pathwatcher/build'

> serialport@1.7.4 install /tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport
> node-pre-gyp install --fallback-to-build --runtime=atom-shell

Unknown Runtime: 'atom-shell'

npm WARN installMany node-pre-gyp was bundled with serialport@1.7.4, but bundled package wasn't found in unpacked tree
npm WARN installMany nopt was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany npmlog was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany request was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany semver was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany tar was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany tar-pack was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany mkdirp was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany rc was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
npm WARN installMany rimraf was bundled with node-pre-gyp@0.6.11, but bundled package wasn't found in unpacked tree
node-pre-gyp ERR! install error 
node-pre-gyp ERR! stack Error: Unknown Runtime: 'atom-shell'
node-pre-gyp ERR! stack     at get_runtime_abi (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:76:19)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:265:19)
node-pre-gyp ERR! stack     at install (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:138:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Linux 3.13.0-65-generic
node-pre-gyp ERR! command "node" "/tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build" "--runtime=atom-shell"
node-pre-gyp ERR! cwd /tmp/apm-install-dir-11592-8050-vmzzde/node_modules/spark-dev/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.11
node-pre-gyp ERR! not ok 
npm ERR! Linux 3.13.0-65-generic
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/deadlock/.atom/.apm/.apmrc" "--userconfig" "/home/deadlock/.atom/.apmrc" "install" "/tmp/d-11592-8050-n5rynl/package.tgz" "--target=0.22.0" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build --runtime=atom-shell`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build --runtime=atom-shell'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build --runtime=atom-shell
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-11592-8050-vmzzde/npm-debug.log
```

newzealandpaul, 2015-10-03:
I am getting a similar error installing the latest version, and like you the log does not exist. I am running Atom 1.0.19.

suda, 2015-10-06:
@newzealandpaul Could you try installing `spark-dev` package again? There were some 3rd party changes that broke it but should be fixed now.

newzealandpaul, 2015-10-06:
@suda, it installs now, and there is a Particle menu item in the OSX menu but, but the menu is empty (nothing pops up when I click it). The console has this error:

2/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:755 Failed to require the main module of 'spark-dev' because it requires an incompatible native module.
Run `apm rebuild` in the package directory to resolve.
index.js:46 Window load time: 1557ms

$ which apm
/usr/local/bin/apm
$ apm rebuild
Rebuilding modules ✓
$

Did not seem to do anything. I get the same error.

suda, 2015-10-07:
@newzealandpaul I added mention in [Installation section](https://github.com/spark/spark-dev#installation) on how to rebuild native modules that cause this behavior.

newzealandpaul, 2015-10-08:
Now it installs, but thows an erorr when I try to Compile in the Cloud ....

**Atom Version**: 1.0.19
**System**: Mac OS X 10.10.5
**Thrown From**: [spark-dev](https://github.com/spark/spark-dev) package, v0.0.27

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, uv_chdir

```
At /Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:477

Error: ENOENT: no such file or directory, uv_chdir
    at Error (native)
    at /Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:477:13
    at Object.module.exports.projectRequired (/Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:255:5)
    at /Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:464:56
    at Object.module.exports.loginRequired (/Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:240:5)
    at Object.module.exports.compileCloud (/Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:464:38)
    at atom-workspace.disposables.add.atom.commands.add.spark-dev:compile-cloud (/Users/paul/.atom/packages/spark-dev/lib/spark-dev.coffee:77:38)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:245:29)
    at CommandRegistry.handleCommandEvent (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:145:19)
    at EventEmitter.<anonymous> (/Applications/Atom.app/Contents/Resources/app.asar/src/window-event-handler.js:70:30)
    at emitOne (events.js:77:13)
    at EventEmitter.emit (events.js:169:7)
```

### Commands

```
     -1:05.6.0 core:move-left (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
  2x -1:05.0 core:backspace (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -1:04.4.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:03.0 core:delete (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -1:01.9.0 core:paste (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -1:00.5.0 core:backspace (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -0:58.9.0 core:save (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -0:51.5.0 spark-dev:flash-cloud (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -0:34.8.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:16.5.0 core:save (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -0:08.2.0 spark-dev:select-device (atom-text-editor.editor.vim-mode.insert-mode.is-focused)
     -0:06.3.0 spark-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:06.3.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:06.3.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:05.1.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:03.5.0 spark-dev:compile-cloud (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "highlight-column",
      "metrics"
    ]
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.28.14
autocomplete-php, v0.3.7
autocomplete-ruby, v0.1.0
highlight-line, v0.11.0
language-particle, v0.3.4
minimap, v4.13.4
particle-dev-cloud-functions, v0.0.9
particle-dev-cloud-variables, v0.0.8
particle-dev-release-notes, v0.53.2
particle-offline-compiler, v0.1.3
project-manager, v2.6.1
spark-dev, v0.0.27
tool-bar, v0.1.9
tree-view-git-status, v0.2.2
vim-mode, v0.60.0
wordpress-api, v1.2.1

# Dev
No dev packages
```

suda, 2015-10-08:
@newzealandpaul this may be caused by opening one file instead of it's parent directory. Try doing **File** -> **Open...** and choose a directory.

newzealandpaul, 2015-10-09:
@suda that fixed it thanks

closed 2015-10-09 by suda