particle-iot-archived/particle-dev #120: Latest readme.md removed the linux install instructions

cdeadlock, 2015-10-06:
Which I would assume are still required, since my latest atom update to 0.0.27 is complaining about serialport driver again

cdeadlock, 2015-10-06:
Hmm I guess this is a different problem now, the old linux install instructions do not fix this:

Atom shows a small bug icon in the bottom right, clicking on it goes to a window that shows
```
The following packages could not be loaded because they contain native modules that aren't compatible with this version of Atom.

Previous Atom versions shipped with Chrome 31 and Node 0.11.10 but Atom now ships with Chrome 41.0.2272.76 and Node 1.5.1.

The packages listed should now ship versions of these native modules that are compatible with Node 1.5.1.

Updates for these packages may already be available that resolve this issue.

If no update is available you may want to notify the package author that their package isn't supported in Atom 1.0.9 because of the Chrome 41.0.2272.76 and Node 1.5.1 upgrade.

If you think a package is listed here and should no longer be, click the button below to reload Atom and recheck all packages.
```

And the problem is
```
Spark Dev 0.0.27
Check for UpdateReport IssueDisable Package
Listed below are the incompatible native modules that this package depends on.

serialport@1.7.4
Error message: Cannot find module '/home/deadlock/.atom/packages/spark-dev/node_modules/serialport/build/serialport/v1.7.4/Release/node-v43-linux-x64/serialport.node'
```

suda, 2015-10-07:
@cdeadlock I added info about Red Bug icon in [installation](https://github.com/spark/spark-dev#installation).

commit 643404c2 (2015-10-07): Add Linux dependencies
Closes #120
closed 2015-10-07 by suda