particle-iot-archived/particle-dev #121: Use linter for error highlighting

steelbrain, 2015-10-10:
Howdy Spark Team! Awesome package you got there, I am the author of the [`linter`](https://atom.io/packages/linter) package and would like to invite you to try it out. Linter supports both pull and push message mechanisms, if you use pull, your linter will be triggered when user saves or stops typing (configurable), in push, you can push messages to linter whenever a command of your package is triggered. Here's some of the benefits of using linter.

 - All of the linter providers's messages are shown in the same place so they won't take a lot of user space
 - It is fast, and gets faster with every release
 - It contains a lot of settings that users can tweak to make it fit their taste, they can enable/disable three filter tabs, error underlining, inline bubbles and gutter decorations along with gutter placement.
 - You can focus on doing your part and leave the rest to linter.

Besides that, linter is well maintained and we always treat requests/suggestions from IDE packages with high priority :wink:. If you say that this way your package users will have to download linter package, well we have written [`atom-package-deps`](https://npmjs.org/package/atom-package-deps) for that. So as soon as somebody installs your package and doesn't have linter, it'll be automatically installed for them.

[Linter API Reference](https://github.com/atom-community/linter/wiki/Linter-API)

Fixes #79
Fixes #77

dmiddlecamp, 2015-10-10:
Awesome, thanks!  We'll check it out :)

suda, 2015-10-12:
@steelbrain thx for this suggestion! We're already testing different linter packages (namely [linter-clang](https://atom.io/packages/linter-clang), [https://atom.io/packages/linter-cpplint](linter-cpplint) and [https://atom.io/packages/linter-gcc](linter-gcc)). There are some problems we encountered like:

- code is compiled against our firmware which declares a lot of constants that are highlighted as errors
- code can be written in C++ and Wiring (.ino files) which would need a preprocessor pass before being linted

We're looking forward for adding linter to our toolchain but it will need some additional work before it's suitable for our users.

steelbrain, 2015-10-12:
@suda I am not sure what you mean when you say that linter needs some additional work. Linter is actually the base. It doesn't know anything about the errors or file itself. All it does is trigger the providers and then highlight the errors that they tell it to.

If you say that you are using the `cpplint` provider and are getting some constant definition errors. Well, If you know what those errors are. You can fork that provider and make it ignore those certain errors. It's as simple as doing `Array.prototype.filter`.

Linter accepts both an array of messages to highlight or a promise that when resolved returns an array of messages to highlight. You can just use the promises, pre-process the file, lint the preprocessed one and then feed those errors to linter.

I hope it makes sense. If it doesn't, let me know and I'll explain it further.

suda, 2015-10-12:
@steelbrain by linter I meant linter provider (sorry for not being clear). Solution you proposed is exactly what we wanted to do :)