particle-iot-archived/particle-dev #123: Empty errors array causes crash

suda, 2015-10-13:
See: https://github.com/spark/particle-dev-cloud-functions/issues/2

closed 2016-04-18 by suda

Aditya90, 2016-08-26:
I am also running into this issue. How am I supposed to resolve this?

suda, 2016-09-02:
@Aditya90 are you using standalone Particle Dev App or Atom and `particle-dev` package?

Aditya90, 2016-09-05:
I am using the standalone particle dev PC application. Also, I am unsure about the "particle-dev" package you are talking about. I am coming back to an older project of mine in the hopes of completing it - https://github.com/Aditya90/GardenWatering-Firmware. I am trying to compile master but am running into problems. It used to compile at one point.

suda, 2016-09-05:
@Aditya90 by `particle-dev` package I mean using [Atom](https://atom.io/) and installing [`particle-dev` package in Atom](https://atom.io/packages/particle-dev). The package contains latest development including bugfixes.

commit 0421b899 (2016-04-18): Fix #123