particle-iot-archived/particle-dev #124: Uncaught Error: ENOENT: no such file or directory, uv_chdir

DazWilkin, 2015-10-15:
I'm unable to use Particle Dev. Compile and [upload|show errors] both return this error:

    Uncaught Error: ENOENT: no such file or directory, uv_chdir

C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:669

"...\resources" exists but contains directories [app, app.asar.unpacked,cli] and files [app.asar,atom.asar,LICENSE.md]

'app.asar.unpacked' contains node_modules\spark-dev\node_modules but no 'lib'

```
Error: ENOENT: no such file or directory, uv_chdir
    at Error (native)
    at C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:669:21
    at Object.module.exports.projectRequired (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:371:14)
    at C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:647:24
    at Object.module.exports.loginRequired (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:355:14)
    at Object.module.exports.compileCloud (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:645:19)
    at atom-workspace.disposables.add.atom.commands.add.spark-dev:compile-cloud (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\node_modules\spark-dev\lib\spark-dev.js:134:26)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\src\command-registry.js:245:29)
    at CommandRegistry.handleCommandEvent (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\[ACCOUNT]\AppData\Local\particle\app-1.0.15\resources\app.asar\src\command-registry.js:145:19)
```

The error message additionally reports:
"The error was thrown from the spark-dev package. Atom is out of date; 1.0.15 installed; 1.0.19 latest. Upgrading to the latest version may fix this issue"

I have Atom installed directly too:
C:\Users\[Account]\AppData\Local\atom

Atom reports version 1.0.19 and contains directories seemingly confirming this [app-1.0.17, app-1.0.19]

Interestingly, Particle is installed:
C:\Users\[Account]\AppData\Local\particle

Particle reports version 1.0.15 and contains a directory confirming this [app-1.0.15]

If I "Help", "Check for Update" in particle, it reports that 1.0.15 is the latest version.

This worked when I first installed it but the updates appear to have broken it.

suda, 2015-10-19:
@DazWilkin just [released 1.0.19](https://github.com/spark/particle-dev-app/releases/tag/v1.0.19). Could you install it and check if this is fixed?

DazWilkin, 2015-10-19:
That worked. Thank you!

closed 2015-10-19 by DazWilkin