particle-iot-archived/particle-dev #125: Uncaught TypeError: Cannot read property 'length' of undefined

crossfireprod, 2015-10-17:
[Enter steps to reproduce below:]

1. <Start w/ Fresh Install of Atom & Particle Dev Related Packages.>
2. Log In
3. "Select device you want to work on"
4. "Setup device's WiFi Credentials"

**Atom Version**: 1.0.19
**System**: Ubuntu 15.04
**Thrown From**: [spark-dev](https://github.com/spark/spark-dev) package, v0.0.31

### Stack Trace

Uncaught TypeError: Cannot read property 'length' of undefined

```
At /home/zach/.atom/packages/spark-dev/lib/utils/serial-helper.coffee:17

TypeError: Cannot read property 'length' of undefined
  at /home/zach/.atom/packages/spark-dev/lib/utils/serial-helper.coffee:17:7
  at /home/zach/.atom/packages/spark-dev/node_modules/serialport/serialport.js:565:11
  at FSReqWrap.oncomplete (fs.js:82:15)

```

### Commands

```
  2x -1:18.4.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:18.4.0 spark-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:18.4.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:18.4.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:10.3.0 spark-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:08.5.0 spark-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:08.5.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:08.5.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:08.1.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:07.2.0 spark-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:05.7.0 core:confirm (atom-text-editor.editor.mini.is-focused)
     -1:05.7.0 spark-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:05.7.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:05.7.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  3x -1:05.5.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:01.0 spark-dev:setup-wifi (atom-text-editor.editor)
```

### Config

```json
{
  "core": {}
}
```

### Installed Packages

```coffee
# User
console-panel, v0.2.0
particle-dev-cloud-functions, v0.0.9
particle-dev-cloud-variables, v0.0.8
spark-dev, v0.0.31
tool-bar, v0.1.9

# Dev
No dev packages
```

crossfireprod, 2015-10-18:
Also occurs when trying to open / use the Serial Monitor.

crossfireprod, 2015-12-01:
@suda Any update on this?  My Photon is sitting lifeless.

suda, 2015-12-02:
@crossfireprod sorry for keep you waiting. This is planned to be fixed this sprint. In the mean time you could try using [CLI setup](https://docs.particle.io/guide/tools-and-features/cli/#particle-setup).

crossfireprod, 2015-12-02:
Sweet!  Thank you, I really appreciate it.

(The Photon seems like an *awesome* device, but I've had nothing but trouble trying to get a reliable IDE setup...)

commit 4c70c397 (2016-04-18): Fix #125
closed 2016-04-18 by suda