particle-iot-archived/particle-dev #127: Uncaught TypeError: Cannot read property '0' of undefined

az-ericso, 2015-10-28:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.0.19
**System**: Mac OS X 10.10.5
**Thrown From**: [spark-dev](https://github.com/spark/spark-dev) package, v0.0.31

### Stack Trace

Uncaught TypeError: Cannot read property '0' of undefined

```
At /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/decorators/unhandledRejection.js:80

TypeError: Cannot read property '0' of undefined
  at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/lib/spark-dev.js:790:58
  at tryCatchReject (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/makePromise.js:845:30)
  at runContinuation1 (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/makePromise.js:804:4)
  at Fulfilled.when (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/makePromise.js:592:4)
  at Pending.run (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/makePromise.js:483:13)
  at Scheduler._drain (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/Scheduler.js:62:19)
  at Scheduler.drain (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/spark-dev/node_modules/when/lib/Scheduler.js:27:9)
  at doNTCallback0 (node.js:416:9)
  at process._tickCallback (node.js:345:13)

```

### Commands

```
     -0:28.4.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling)
     -0:28.2.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:28.2.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:28.1.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:28.1.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:28.1.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:28.1.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:23.2.0 application:open-file (atom-text-editor.editor)
     -0:04.1.0 spark-dev:select-device (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:03.1.0 spark-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:03.1.0 spark-dev:update-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:03.1.0 spark-dev:append-menu (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:02.6.0 spark-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:01.1.0 spark-dev:flash-cloud (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
     -0:01.0 spark-dev:update-compile-status (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```

suda, 2015-10-29:
@az-ericso what are the steps to reproduce this issue?

az-ericso, 2015-10-29:
@suda sorry for not being in details.  I just updated the description above.

puneetteja, 2015-11-20:
I am getting the same error and here is what I did to get rid of it!!!
The error appears if I open the source files in a sub-folder like this:
[image]
Note the sub-folders under the main folder: "9_All_params_updated". My code is in the sub-folder "SourceFilesWONeoPixelWOSDCard"

Now if I close Particle Dev and reopen directly the folder "SourceFilesWONeoPixelWOSDCard" as shown here:
[image]
I am able to compile EXACTLY the same code for both Photon & Core.

So the problem is probably the code & files in other folders. Apparently Particle Dev sends all the sub-folders to Particle for compilation whereas it "should" send only the open folder. Of course, a case can be made for  sending all sub-folders as well. Enjoy!

suda, 2015-11-20:
That is correct, Dev will send whole project to be compiled. Implementing #39 could help specify what to include. But still exception above should be caught and shown in more human readable format.

bitwisekaizen, 2015-11-22:
I'm running into the same issue when trying to integrate httpsclient-particle with Particle Dev.  @puneetteja's workaround did not work for me as it probably has something to do with the httpsclient-particle library specifically.

To reproduce:
1. Clone the httpsclient-particle repository.
2. Remove everything with the exception of .c, .h, and .cpp files.
3. Add the resulting source as a project folder in Particle Dev.
4. Try to compile the source in the cloud.

Expected: Source compiles.
Results: This bug is referenced and the "compiling in the cloud..." spinning wheel never stops spinning.

Jerware, 2016-01-26:
@suda I appear to get this error from Spark Dev when my source code reaches an undetermined character threshold. I started hitting it after adding some functions. After trying to debug the functions, I realized that if I simply removed them and added about one hundred lines of white space to my source that it threw the same error.

For the record, I have about 3500 lines (3600+ gets this error). This is in addition to four libraries, three of them considerably large (FastLED, SdFat, and WebServer).

I'd be curious to learn if I get the same error with a local toolchain but I haven't had the time to try to set that up.

suda, 2016-01-26:
@Jerware that's interesting. What is result of CLI's `compile` command?

Jerware, 2016-01-26:
I've never successfully compiled from the CLI. I currently get:

Compile failed. Exiting.
undefined

I sent a zip of my source to the email on your profile, but I suspect a maximum payload size is at the root of this error so it's not my particular source code... just the amount of it. Just add a couple hundred lines to mine and it should throw the error.