particle-iot-archived/particle-dev #130: spark-dev install failure

matt-gordonboyworks, 2015-11-17:
Atom 1.2.1, OS X El Capitan 10.11.1, there's a failure to install (no red bug indicating the option to rebuild modules.)  Path  /private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/npm-debug.log does not exist.

Matthews-iMac:~ matt$ apm install spark-dev
Installing spark-dev to /Users/matt/.atom/packages ✗

> serialport@1.7.4 install /private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/node_modules/spark-dev/node_modules/serialport
> node-pre-gyp install --fallback-to-build --runtime=atom-shell

  CXX(target) Release/obj.target/serialport/src/serialport.o
Failed to execute 'node-gyp build --fallback-to-build --runtime=atom-shell --module=/private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/node_modules/spark-dev/node_modules/serialport/build/serialport/v1.7.4/Release/electron-v0.34-darwin-x64/serialport.node --module_name=serialport --module_path=/private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/node_modules/spark-dev/node_modules/serialport/build/serialport/v1.7.4/Release/electron-v0.34-darwin-x64' (1)

npm WARN installMany node-pre-gyp was bundled with serialport@1.7.4, but bundled package wasn't found in unpacked tree
npm WARN installMany nopt was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany npmlog was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany request was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany semver was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany tar was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany tar-pack was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany mkdirp was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany rc was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany rimraf was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:41:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:53:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:38:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:46:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/matt/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:326:13: note: candidate template ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:683:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
In file included from ../src/serialport.cpp:3:
[... 83 lines cut ...]
node-pre-gyp ERR! command "node" "/private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/node_modules/spark-dev/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build" "--runtime=atom-shell"
node-pre-gyp ERR! cwd /private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/node_modules/spark-dev/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.12
node-pre-gyp ERR! not ok 
npm ERR! Darwin 15.0.0
npm ERR! argv "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/matt/.atom/.apm/.apmrc" "--userconfig" "/Users/matt/.atom/.apmrc" "install" "/private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/d-1151017-5672-1nzax7m/package.tgz" "--target=0.34.0" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build --runtime=atom-shell`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build --runtime=atom-shell'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build --runtime=atom-shell
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/mv/p6jbtd3j20g3_dtfv4vz2jph0000gn/T/apm-install-dir-1151017-5672-1wuu87o/npm-debug.log

suda, 2015-11-17:
Thx Matt, will look into this.

tlorens, 2015-11-19:
Fresh install/reinstall of OS X. - Atom 1.2.1 / OS X Yosemite (10.10.5).  I've got the same problem.  Just to rule out El Capitan (10.11.1).

tlorens, 2015-11-25:
Appears to be a problem after Atom 1.1.0.  Installed 1.2.x (bets included) and continues to fail.

drankinn, 2016-01-04:
it's working good for me now. 
Thanks!

closed 2016-03-10 by suda