particle-iot-archived/particle-dev #131: serialport problems, el capitan (10.11.1)

hwestbrook, 2015-11-19:
Received the error below when trying to install the spark-dev atom package. 

Using el capitan 10.11.1, with Atom 1.2.3 and Node 5.0.0.

Error: 
```
> serialport@1.7.4 install /private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/apm-install-dir-1151019-9254-3xnw6a/node_modules/spark-dev/node_modules/serialport
> node-pre-gyp install --fallback-to-build --runtime=atom-shell

  CXX(target) Release/obj.target/serialport/src/serialport.o
Failed to execute 'node-gyp build --fallback-to-build --runtime=atom-shell --module=/private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/apm-install-dir-1151019-9254-3xnw6a/node_modules/spark-dev/node_modules/serialport/build/serialport/v1.7.4/Release/electron-v0.34-darwin-x64/serialport.node --module_name=serialport --module_path=/private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/apm-install-dir-1151019-9254-3xnw6a/node_modules/spark-dev/node_modules/serialport/build/serialport/v1.7.4/Release/electron-v0.34-darwin-x64' (1)

npm WARN installMany node-pre-gyp was bundled with serialport@1.7.4, but bundled package wasn't found in unpacked tree
npm WARN installMany nopt was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany npmlog was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany request was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany semver was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany tar was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany tar-pack was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany mkdirp was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany rc was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
npm WARN installMany rimraf was bundled with node-pre-gyp@0.6.12, but bundled package wasn't found in unpacked tree
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:41:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:53:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:38:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/src/node_buffer.h:46:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/hwestbrook/.atom/.node-gyp/.node-gyp/0.34.0/deps/v8/include/v8.h:326:13: note: candidate template ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:683:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
[... 85 lines cut ...]
node-pre-gyp ERR! cwd /private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/apm-install-dir-1151019-9254-3xnw6a/node_modules/spark-dev/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.12
node-pre-gyp ERR! not ok
npm ERR! Darwin 15.0.0
npm ERR! argv "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/hwestbrook/.atom/.apm/.apmrc" "--userconfig" "/Users/hwestbrook/.atom/.apmrc" "install" "/private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/d-1151019-9254-1qagqxg/package.tgz" "--target=0.34.0" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build --runtime=atom-shell`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build --runtime=atom-shell'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build --runtime=atom-shell
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/hm/nkjp28h914ddhwb8jz_5nq400000gn/T/apm-install-dir-1151019-9254-3xnw6a/npm-debug.log
```

hwestbrook, 2015-11-19:
Doing some more testing, I no longer experience this issue if I downgrade atom back to v1.1. Is this related to >=1.2?

suda, 2015-11-20:
@hwestbrook It is possible. I will do some tests against it.

aleemstreak, 2015-12-28:
Any luck on finding this? Likely the uptick in reports of this issue is due to a bunch of people getting a particle for christmas :)

suda, 2016-01-04:
@hwestbrook & @aleemstreak can you check again now? I updated dependent package which should fix the problem.

closed 2016-01-04 by suda