particle-iot-archived/particle-dev #138: Code edits not reflected in firmware load via flash command

uchobby, 2016-03-29:
When I edit code then tap the flash to load it into my device, everything looks normal, the device restarts etc after being flashed etc.. But the running code is a prior version, not the one I just edited.

The fix I've found is to select a different device, then re-select the desired target, then flash...

This is either a bug in the system of some flaw in my development setup. How can I help fix this?

suda, 2016-03-29:
What's the device you're trying to flash (Core/Photon/Electron)?
Does this happen every time? (can you reproduce steps you mentioned each time)
Could you try using [CLI](https://github.com/spark/particle-cli#installing) to start `particle subscribe mine` and see if it shows `spark/device/app-hash` event after flashing from Dev?

uchobby, 2016-03-29:
Is a Photon.  Does not happen ever time. I don't yet have a correlation to an action so that I can repeat the issue. When it does happen, everything seems to work but it's loading previous code. I include a version ID which is output on the LEDs of the Internet Button so I can tell that it's updated.

When it starts this evil issue, I end up trying many things and sometimes it starts working again. Things like selecting a different Photon, switching projects, close and re-open Dev etc.. Sometimes one of these fixes other times not.  I am thinking there is an issue with the cloud compile/save process. 

I've even put in errors in the code to see if they are caught. I will have to re-do this experiment as I'm only 90% sure the errors were not caught on recompile, proving in some way that it's a cloud issue.

Is there a way I can see more information about the cloud process? Like when a file is sent up, when it's compiled, etc...  

I did have a initialization issue, calling one of the Accel methods in the InternetButton lib before init was called.  That was crashing my button completely, it's possible that some of my problems were related to that.  Some, but not all.

Last night, I had to move back to the online IDE to continue working as Dev was not updating as I edited.

I am a 35+ year experienced embedded developer and love this platform. I am willing to help figure this out anyway I can.  I will try the CLI suggestion and get back.

uchobby, 2016-03-29:
OK, output from CLI. Made a single code change, just bumped my version ID, in the editor and simply clicked flash.  Process seemed perfectly normal, device LEDs flashed as if being programmed, serial output showed reset process but version ID did not change.
{"name":"spark/flash/status","data":"started ","ttl":"60","published_at":"2016-03-29T19:25:56.042Z","coreid":"I removed 24 numbers here for privacy"}
{"name":"spark/flash/status","data":"failed ","ttl":"60","published_at":"2016-03-29T19:26:03.667Z","coreid":"I removed 24 numbers here for privacy"}
{"name":"spark/status","data":"online","ttl":"60","published_at":"2016-03-29T19:26:03.665Z","coreid":"I removed 24 numbers here for privacy"}

uchobby, 2016-03-29:
Without any edits, I tried several different things, "compile in the cloud" followed by "flash", forced save of file, reselected the device etc.. each time I tried and got the old version of code on flashing.  Then I got this and the new version ran.
{"name":"spark/flash/status","data":"started ","ttl":"60","published_at":"2016-03-29T19:33:23.272Z","coreid":"2c0023000747343232363230"}
{"name":"spark/flash/status","data":"failed ","ttl":"60","published_at":"2016-03-29T19:33:31.536Z","coreid":"2c0023000747343232363230"}
{"name":"spark/status","data":"online","ttl":"60","published_at":"2016-03-29T19:33:31.545Z","coreid":"2c0023000747343232363230"}
{"name":"spark/device/app-hash","data":"EB0C292754EFB9344FA5D30FACAF7AFD64AFBCD2D14D2A5C1BAAC5CC59A9204D","ttl":"60","published_at":"2016-03-29T19:33:32.936Z","coreid":"2c0023000747343232363230"}

What is the failed error above. I see the "failed" message on every flash, good or bad.

uchobby, 2016-03-29:
Maybe a clue/complication... The editor does not seem to save edits before doing a cloud compile. I've seen several cases where the error reported was corrected but did not clear on compile until after a file save.

suda, 2016-03-30:
One of the reasons might be that [Dev doesn't automatically save the file you're working on](https://github.com/spark/spark-dev/issues/31) so you have to do it manually before each compile.

`spark/flash/status` failed is a false negative. If you see `spark/device/app-hash` this means flash has succeeded and an app has been updated.

Also Flash button will flash latest `.bin` file in the directory if it exists (instead of compiling latest version) so make sure you don't have any `photon_firmware_*.bin` files there.

Let me know if this helped.

uchobby, 2016-03-31:
The "failed" message should be listed as a bug.

Are you saying that I must manually delete the .bin file after each compile?  The dev process is..
1) Edit your file.
2) Save your file.
3) Delete any previously generated .bin files
4) Flash

Why can't the development tool do steps 2, 3 ,and 4?  Every other IDE does...

embedded-creations, 2016-04-01:
I agree with @uchobby, I'm finding it too easy to accidentally compile an unsaved sketch, or flash an old version of firmware.

Also I noticed this behavior which I plan to report as a new issue (if it's not already an issue), it might help here.  If you restart Dev, no devices are selected, and when you press compile, it compiles firmware for the Core.  If you then select a Photon as your device and flash, it will flash with the previous photon_firmware.bin that was in the directory, and doesn't recompile.  If there was no photon_firmware.bin, then it does appear to compile fresh.

suda, 2016-04-01:
There are two options in settings that control this behaviour:

[image]

By default flash will try to use existing binary. This is intentional as it could be build using CLI, local compiler or downloaded from Build. I think another option: "Always compile new firmware when flashing" would be helpful in mentioned situations.

uchobby, 2016-04-01:
The CLI would be used by advanced users so I would focus Dev on the Arduino crowd.  Make it hard for things to fail rather then hard to get things to work.  I totally understand trying to be versatile but it's not a good trade off to benefit 10% of your users at the expense of 90%.

I use the CLI now with my normal code editor, this has been reliable for me so far.

If you need any testing done with the Dev tool, I'd be happy to switch back.

suda, 2017-02-14:
Dev now saves opened editors before compilation and has more sensible platform selection. This should help with unexpected behaviour.

closed 2017-02-14 by suda