particle-iot-archived/particle-dev #140: "Compile in cloud" treats include paths differently than web IDE, breaks most of the library code examples

DaveCasey, 2016-04-07:
Most library example code uses includes of the form: #include "LibraryName/LibraryName.h"

This is also the form that the web IDE uses when it adds a library for you:
// This #include statement was automatically added by the Particle IDE.
#include "OneWire/OneWire.h"

The dev environment requires you to strip the foldername from the include:
#include "OneWire.h"
will compile.
#include "OneWire/OneWire.h"
will fail with the following error:

[image]

Regardless of whether you use this directory structure:
[image]

or this directory structure:
[image]

DaveCasey, 2016-04-07:
Note that the following block from the spark-dallas-temperature library's spark-dallas-temperature.h
must also be updated in order to get Dev's "Compile in the cloud" to succeed.

#elif defined(SPARK) or defined(STM32F10X_MD)
    #include "application.h"
    #include "../OneWire/OneWire.h"
#endif

must be changed to:
#elif defined(SPARK) or defined(STM32F10X_MD)
    #include "application.h"
    #include "OneWire.h"
#endif

Asking users to modify libraries that they don't own seems like a bad idea....

suda, 2016-04-08:
Hi Dave, this is [an open issue](https://github.com/spark/spark-dev/issues/71) with our dev tools, API and compile server but we'll keep doing incremental fixes so all the parts to fix this problem will be in place.

closed 2016-04-08 by suda