particle-iot-archived/particle-dev #143: Add firmware version targeting

suda, 2016-04-13:


fourcube, 2016-04-15:
Is there any way to force this right now?

suda, 2016-04-15:
@fourcube not really. Only way would be to use CLI for compilation

fourcube, 2016-04-21:
@suda thank you. I've actually found a way to force the version by hacking the spark-dev -> spark-api package to supply the firmware version in the compilation HTTP request. 

Compilation worked fine, but the behaviour of my application changed and stopped working. I'll just wait a little :+1:

csoulakian, 2016-10-20:
@fourcube Where exactly were you able to insert the target arg to compile? Is there a single place in the code that uses the CLI commands that I can simply add the target for compiling and flashing OTA?

fourcube, 2016-10-21:
@csoulakian It looks like some repositories and modules were renamed over the course of the year. I currently do not have access to the dev machine on which I've built my particle project. Anyways, what I did was to directly edit the source of one of the spark/particle Atom packages which were installed on my local machine. 

After you have figured out the correct version string of the framework, I *think* that you can still compile against that specific version by manipulating:

https://github.com/spark/particle-api-js/blob/master/src/Particle.js#L199 (compile and flash)
or
https://github.com/spark/particle-api-js/blob/master/src/Particle.js#L230 (compile only)

csoulakian, 2016-11-02:
As a work around, I'm using [process-palette](https://atom.io/packages/process-palette). It's really simple and allows you to create input dialogs and variables which I'm using for the device name and firmware version. They conveniently pop up when you run the command and you have to put the CLI commands in yourself when you set it up, but it's almost as good as the built in features. All I really use from the built in features now are the cloud functions/variables, serial monitor, and autocomplete. I compile and flash using process-palette.

suda, 2017-10-02:
This is available since [v0.1.20](https://github.com/spark/particle-dev/releases/tag/v0.1.20) :)

closed 2017-10-02 by suda