particle-iot-archived/particle-dev #149: build didn't produce binary Error: Command Failed

hwbrill, 2016-06-06:
This indication shows after I select 'compile in cloud and show any errors if any' button.  There were no compile errors.

suda, 2016-06-07:
@hwbrill can you share a screen shot of Dev right after it fails?

hwbrill, 2016-06-07:
Prior to 'Compile to cloud and show any errrors'
[image]

After the 'Compile to cloud and show any errors'
[image]

suda, 2016-06-07:
It looks like you have the whole library opened in Dev which it will try to compile as it doesn't compile just the opened tab but the whole directory. Currently [there is no direct way to use libraries](https://github.com/spark/particle-dev/issues/71) in Dev and you would need to [follow these instructions](https://docs.particle.io/guide/tools-and-features/dev/#using-community-libraries) to use them.

hwbrill, 2016-06-07:
I've also have been fooling around with my library's file structure and found that each code file, *.ino, must be in a folder by it's self for the cloud compile to succeed.  I knew about the firmware, to be in same directory as the *.ino file but, I didn't see anything about not having other code files in the same directory during compile...

closed 2016-06-07 by hwbrill