particle-iot-archived/particle-dev #151: Installing this package on a fresh atom installation results in an error

htko89, 2016-07-06:
I'm encountering an error when installing "particle-dev-complete" or "particle-dev" on a fresh atom installation (1.8.0)

I'm not savvy enough to figure out the error, but I'm wondering if there is anyone here that can help?

Here is the error log: http://pastebin.com/dCrGkreV

I've also posted this on the particle dev complete github ( https://github.com/spark/particle-dev-complete/issues/1 ) but now I see that this is more of a particle-dev error

Windows 10 x64

suda, 2016-07-07:
Hi Tony,

it's an issue with `serialport` package that Partice Dev uses. I would recommend to install a precompiled [Particle Dev App](https://www.particle.io/dev) that has all the prerequisites.

If you still want to try installing Particle Dev package, you would need to install [node-gyp dependencies](https://github.com/nodejs/node-gyp#installation) first.

htko89, 2016-07-09:
Hey

Thanks for the reply. I would like to install it to Atom and I have installed node-gyp along with node. It is still throwing the error. Am I doing something wrong?

Thanks

suda, 2016-07-11:
@htko89 Did you install all the dependencies like Visual Studio and Python?

htko89, 2016-07-14:
Yes I have, sorry for the late reply. Is there any dependencies within visual studio community that need to be installed ? (I have picked just the default install preset)

htko89, 2016-07-14:
I do have a question: It's not clear on whether or not I have to install node.js separately on my system (which I have) and where to install node-gyp. Sorry if this makes no sense, but I imagine atom is using its own version of node (from what I'm reading in the error log, the versions don't match up)

If I'm right, or partially on track, what are the instructions to set up node-gyp correctly?

suda, 2016-07-14:
You're exactly right. Atom bundles its own version of Node. In your logs I see following error: 
```
error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". 
```
So I guess installing the SDK should help

pingswept, 2016-08-10:
Hey, I think I'm running into either the same error, or a very similar one.

Full error message: http://pastebin.com/hmuB9Eh3

I was trying to install Particle Dev on a fresh build of Atom from source on Ubuntu 16.04.

➜  ~ atom-beta --version
Atom    : 1.10.0-beta4
Electron: 0.37.8
Chrome  : 49.0.2623.75
Node    : 5.10.0

(As recommended above, I'll try installing a prebuilt package, but there's no x86 .deb available for Atom, so that will mean moving to my laptop and OS X.)

Magnus512, 2016-08-18:
I have the same error in Ubuntu and Fedora

suda, 2016-08-19:
Could you all try [this](https://github.com/spark/particle-dev/issues/152#issuecomment-239474196)? It might help

Magnus512, 2016-08-19:
As soon as I can

xavierdebriey, 2016-08-19:
I've also the same problem. I tried your solution, @suda, i've managed to install particle-dev package but the same errors come back when i rebuild the package.

Thank you very much

Magnus512, 2016-08-19:
@suda same error
node-pre-gyp ERR! install error 
node-pre-gyp ERR! stack Error: Unsupported target version: 0.37.8
node-pre-gyp ERR! stack     at get_runtime_abi (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Linux 4.7.1-040701-generic
node-pre-gyp ERR! command "node" "/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/marcus/.atom/packages/particle-dev/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 

npm ERR! Linux 4.7.1-040701-generic
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/marcus/.atom/.apm/.apmrc" "--userconfig" "/home/marcus/.atom/.apmrc" "rebuild" "--target=0.37.8" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE
npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/marcus/.atom/packages/particle-dev/npm-debug.log

suda, 2016-08-22:
Rebuilding will fail as Atom doesn't currently pass `npm_config_runtime`. Removing package and installing it via terminal should be enough to have it working (without rebuilding).

suda, 2017-01-20:
Latest Atom which bundles with latest `apm` should build those packages correctly.

closed 2017-01-20 by suda

GonzRon, 2017-03-21:
@suda im having the same serialport issue when installing particle-dev-complete in the latest atom editor