particle-iot-archived/particle-dev #152: Latest Atom will not Rebuild Package

kleerkoat, 2016-08-01:
getting the following when trying to rebuild package with Atom 1.9.0

2016-08-01 15:32:03.036 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/QuickShowInFinder.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.037 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/QuickFind.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.037 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/OTOpenInITerm.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.038 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/Open With Application.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.039 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/OMQuickHelp.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.039 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/Multiplex.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.040 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/EditorConfig.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.041 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/CodePilot3.xcplugin' not present in DVTPlugInCompatibilityUUIDs
2016-08-01 15:32:03.041 xcodebuild[40286:508259] [MT] PluginLoading: Required plug-in compatibility UUID 1637F4D5-0B27-416B-A78D-498965D64877 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/Alcatraz.xcplugin' not present in DVTPlugInCompatibilityUUIDs
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:590:20: error: no type named 'GCEpilogueCallback' in 'v8::Isolate'
      v8::Isolate::GCEpilogueCallback callback
      ~~~~~~~~~~~~~^
../node_modules/nan/nan.h:596:20: error: no type named 'GCEpilogueCallback' in 'v8::Isolate'
      v8::Isolate::GCEpilogueCallback callback) {
      ~~~~~~~~~~~~~^

suda, 2016-08-02:
It looks like `node-serialport` would require an update. Thx!

suda, 2016-08-12:
The core error here is `apm` bundled with Atom. Current Atom version (1.9.8) is using older apm (1.10.0). There is a fix in [apm 1.12.6](https://github.com/atom/apm/commit/a5d467635d19a0f6ed998d1a5749f50121522a79) but even Atom beta (1.10.0-beta5) is using slightly older version. So I hope this will be fixed in newer Atom beta.

As for temporary workaround you can install the package using command line:
```
export npm_config_runtime=electron
apm install particle-dev
```

Magnus512, 2016-08-19:
I got this 
node-pre-gyp ERR! install error 
node-pre-gyp ERR! stack Error: Unsupported target version: 0.37.8
node-pre-gyp ERR! stack     at get_runtime_abi (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Linux 4.7.1-040701-generic
node-pre-gyp ERR! command "node" "/home/marcus/.atom/packages/particle-dev/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/marcus/.atom/packages/particle-dev/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 

npm ERR! Linux 4.7.1-040701-generic
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/marcus/.atom/.apm/.apmrc" "--userconfig" "/home/marcus/.atom/.apmrc" "rebuild" "--target=0.37.8" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE
npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/marcus/.atom/packages/particle-dev/npm-debug.log

suda, 2016-08-22:
@Magnus512 I tried this on my Elementary (Ubuntu) VM and seems to work. Have you run both `export` and `apm install` in terminal?

Magnus512, 2016-08-22:
@suda yes, well, I'm going to try again removing all packages,

Magnus512, 2016-08-22:
[image]
Everything seems to be ok. but when I open atom I get something like this
[image]
[image]

Jajajaja I got this now :+1:  thanks

kleerkoat, 2016-09-22:
@Magnus512 OT with this post, what status line plugin are you using in zsh?

Magnus512, 2016-09-27:
@kleerkoat itś powerline :)

closed 2017-01-20 by suda