particle-iot-archived/particle-dev #156: ATOM Error immediately after installing particle-dev ver 0.1.3

a33nitsuj, 2016-09-04:
Error compiling Less stylesheet:
.../.atom/packages/particle-dev/styles/particle-dev.less
Line number: 8
'utilities.less' wasn't found
Failed to load the particle-dev package
'utilities.less' wasn't found in /home/justin/.atom/packages/particle-dev/styles/particle-dev.less:8:0
Hide Stack Trace
LessError: 'utilities.less' wasn't found
  at /home/<user>/.atom/packages/particle-dev/styles/particle-dev.less:8:0
The error was thrown from the particle-dev package. You can help by creating an issue. Please explain what actions triggered this error.
This occurred immediately after installing particle-dev ver 0.1.3

kleerkoat, 2016-09-04:
same here...

`**Atom Version**: 1.10.0
**Electron Version**: 0.37.8
**System**: Mac OS X 10.12
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.3

### Stack Trace

Failed to load the particle-dev package

```
At 'utilities.less' wasn't found in /Users/rgilmore/.atom/packages/particle-dev/styles/particle-dev.less:8:0

LessError: 'utilities.less' wasn't found
  at /Users/rgilmore/.atom/packages/particle-dev/styles/particle-dev.less:8:0`

suda, 2016-09-05:
Oh it looks like One theme doesn't have `utilities.less` file. You can avoid this error by switching to Atom  Dark/Light theme for now.

kleerkoat, 2016-09-16:
is this an issue with particle-dev or the atom one themes? should it be reported to them instead?

suda, 2016-09-19:
It's particle-dev issue. We need to remove the dependency on the `utilities.less` file.

brencerddwr, 2016-09-21:
I was having this problem and changed the theme in Atom from "One Dark"(default) to "Atom Dark"(mentioned in particle-dev.less file) and it installed without a problem.

commit cb3d73d5 (2016-09-23): Fix #156 and #157
closed 2016-09-23 by suda