particle-iot-archived/particle-dev #157: Failed to load the particle-dev package

edmundchaniii, 2016-09-11:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.10.2
**Electron Version**: 0.37.8
**System**: Ubuntu 16.04.1
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.3

### Stack Trace

Failed to load the particle-dev package

```
At 'utilities.less' wasn't found in /home/edmund/.atom/packages/particle-dev/styles/particle-dev.less:8:0

LessError: 'utilities.less' wasn't found
  at /home/edmund/.atom/packages/particle-dev/styles/particle-dev.less:8:0
```

### Commands

```
     -0:00.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:00.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:00.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.9.4 (active)
build, v0.65.0 (active)
busy, v0.6.0 (active)
language-ini, v1.16.0 (active)
linter, v1.11.16 (active)
linter-gcc, v0.6.15 (active)
particle-dev, v0.1.3 (inactive)
particle-dev-complete, v1.0.0 (active)
platformio-ide, v1.5.0 (active)
platformio-ide-terminal, v2.2.0 (active)
tool-bar, v1.0.1 (active)
atom-dark-syntax, v0.27.0 (inactive)
atom-dark-ui, v0.52.0 (inactive)
atom-light-syntax, v0.28.0 (inactive)
atom-light-ui, v0.44.0 (inactive)
base16-tomorrow-dark-theme, v1.1.0 (inactive)
base16-tomorrow-light-theme, v1.1.1 (inactive)
one-dark-ui, v1.5.0 (active)
one-light-ui, v1.5.0 (inactive)
one-dark-syntax, v1.3.0 (active)
one-light-syntax, v1.3.0 (inactive)
solarized-dark-syntax, v1.0.2 (inactive)
solarized-light-syntax, v1.0.2 (inactive)
about, v1.5.3 (active)
archive-view, v0.61.1 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.11.2 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.31.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.27.0 (inactive)
autosave, v0.23.1 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.41.0 (active)
bracket-matcher, v0.82.1 (active)
command-palette, v0.38.0 (inactive)
deprecation-cop, v0.54.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.39.0 (active)
find-and-replace, v0.201.0 (inactive)
fuzzy-finder, v1.3.0 (active)
git-diff, v1.1.0 (active)
go-to-line, v0.31.0 (inactive)
grammar-selector, v0.48.1 (active)
image-view, v0.58.2 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.0 (active)
link, v0.31.1 (inactive)
markdown-preview, v0.158.0 (active)
metrics, v0.53.1 (active)
notifications, v0.65.0 (active)
open-on-github, v1.2.0 (inactive)
package-generator, v1.0.0 (inactive)
settings-view, v0.242.2 (active)
snippets, v1.0.2 (active)
spell-check, v0.67.1 (active)
status-bar, v1.4.1 (active)
styleguide, v0.47.0 (active)
symbols-view, v0.113.0 (inactive)
tabs, v0.100.2 (active)
timecop, v0.33.2 (active)
tree-view, v0.208.2 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.34.0 (active)
whitespace, v0.33.0 (active)
wrap-guide, v0.38.1 (active)
language-c, v0.52.1 (active)
language-clojure, v0.21.0 (active)
language-coffee-script, v0.47.2 (active)
language-csharp, v0.12.1 (active)
language-css, v0.37.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.42.1 (active)
language-html, v0.45.1 (active)
language-hyperlink, v0.16.0 (active)
language-java, v0.23.0 (active)
language-javascript, v0.119.0 (active)
language-json, v0.18.2 (active)
language-less, v0.29.5 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.35.0 (active)
language-php, v0.37.2 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.0 (active)
language-ruby, v0.69.0 (active)
language-ruby-on-rails, v0.25.0 (active)
language-sass, v0.56.0 (active)
language-shellscript, v0.22.4 (active)
language-source, v0.9.0 (active)
language-sql, v0.23.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.28.0 (active)
language-toml, v0.18.0 (active)
language-xml, v0.34.9 (active)
language-yaml, v0.26.0 (active)

# Dev
No dev packages
```

cloudzombie, 2016-09-11:
Mine is giving me this 

```
LessError: 'utilities.less' wasn't found
  at /home/josh/.atom/packages/particle-dev/styles/particle-dev.less:8:0
```

Seems like it is specific to a less package that was in particle-dev and is no longer bundled?

kleerkoat, 2016-09-13:
same error here.

mschu2000, 2016-09-15:
and here

davemclaughlin, 2016-09-16:
Have tried so many times to install this but getting nowhere fast with this error.

Is this file available somewhere to download?

davemclaughlin, 2016-09-16:
If you edit line 8 to comment it out in particle-dev.less then it will run without causing the error.

kleerkoat, 2016-09-16:
here's a comment left on issue 156
https://github.com/spark/particle-dev/issues/156

brencerddwr, 2016-09-21:
I was having this problem and changed the theme in Atom from "One Dark"(default) to "Atom Dark"(mentioned in particle-dev.less file) and it installed without a problem.

suda, 2016-09-23:
Closed in cb3d73d

closed 2016-09-23 by suda