particle-iot-archived/particle-dev #164: Particle Dev compiled bin does not work

innoagg, 2016-11-24:
Using a sample blinky app. If I compile with Particle Dev desktop IDE the code will not run on an Electron. It goes to safe mode. Electron has latest .5.3 firmware. If I copy the code to the web IDE Particle Builder and compile it works fine. 

I was working with the desktop IDE just fine and then halfway through the day yesterday the compiled bin files stopped working. Now I have to copy code to web IDE to compile. Rather time consuming.

suda, 2016-11-25:
We've just released 0.6.0 firmware and Particle Dev doesn't have the same failsafe's as Build IDE that would compile against a version running on a device instead of latest one. We'll implement this asap, probably along #143. Sorry for the trouble!

innoagg, 2016-11-28:
I ran particle update through the CLI and the latest version is still 
0.5.3. How do I get the Electron to the latest FW 0.6.0?

<author@noreply.github.com>
Sent: 11/25/2016 6:18:32 AM
work (#164)

innoagg, 2016-11-28:
Sorry, Never mind, I had not update the CLI.

-AP

closed 2017-01-20 by suda