particle-iot-archived/particle-dev #172: Case Sensitivity 0.1.13

magnum129, 2016-12-25:
I'm on Fedora with Atom 1.12.7 and Particle Dev 0.1.13, I keep getting errors about missing libraries when attempting to cloud compile. It appears since the latest update everything is forced to lowercase:

The path the files looks more like: /home/joeUser/Desktop/900MHZ_Particle_Weatherstation/OneWire/OneWire.h

But instead this seems to be where it's looking:
/home/joeuser/desktop/900mhz_particle_weatherstation/onewire/onewire.h'

More stacktrace:

900mhz_particle_weatherstation
onewire
RFM69-Particle
spark-dallas-temperature
SparkFun_Photon_Weather_Shield_Library
weatherStation.ino
weatherStation.ino
1
1

 
ClearConsole
weatherStation.ino1:1TheOutdoorWeatherManPhoton
LFUTF-8C++
Uncaught Error: ENOENT: no such file or directory, stat '/home/joeuser/desktop/900mhz_particle_weatherstation/onewire/onewire.h'
/home/joeUser/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1233
Hide Stack Trace
Error: ENOENT: no such file or directory, stat '/home/joeUser/desktop/900mhz_particle_weatherstation/onewire/onewire.h'
    at Error (native)
    at fs.statSync (fs.js:992:18)
    at Object.fs.statSync (ELECTRON_ASAR.js:275:16)
    at normalizeAndSplitPath (/home/joeUser/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1168:16)
    at Object.pathsCommonPrefix (/home/joeUser/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1220:17)
    at Object.mapCommonPrefix (/home/joeUser/.atom/packages/particle-dev/lib/main.coffee:596:20)
    at /home/joeUser/.atom/packages/particle-dev/lib/main.coffee:630:41
    at Object.loginRequired (/home/joeUser/.atom/packages/particle-dev/lib/main.coffee:300:5)
    at Object.compileCloud (/home/joeUser/.atom/packages/particle-dev/lib/main.coffee:604:65)
    at atom-workspace.<anonymous> (/home/joeUser/.atom/packages/particle-dev/lib/main.coffee:83:55)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/usr/share/atom/resources/app.asar/src/command-registry.js:160:19)
    at executeCallback (/home/joeUser/.atom/packages/tool-bar/lib/tool-bar-button-view.js:105:19)
    at ToolBarButtonView._onClick (/home/joeUser/.atom/packages/tool-bar/lib/tool-bar-button-view.js:69:7)
The error was thrown from the particle-dev package. You can help by creating an issue. Please explain what actions triggered this error.
Create issue on the particle-dev package

magnum129, 2016-12-25:
If I rename all of the includes to lowercase it seems to be happy, despite the includes that actually require each other and have the expected path name in the .h file... Like the OneWire and dallas temp probe...

ybizeul, 2017-01-01:
I'm surprised this bug isn't getting more attention. Can't compile anything since some update.
[image]

ybizeul, 2017-01-01:
edit your `~/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js`

Line 1166, remove the lowerCase call, to make it like this :
```
	var abs = path.resolve(cwd, p); // this might be a bit heavy handed on POSIX...
```

Yeah, you bet it might be __heavy handed on POSIX...__

magnum129, 2017-01-01:
Thanks @ybizeul, that works. I'm guessing most people are using the cloud version, haven't upgraded, or are not on  *nix... and haven't noticed.

ybizeul, 2017-01-01:
thats weird indeed. Maybe we should switch to local compiler!

ybizeul, 2017-01-01:
https://github.com/spark/particle-library-manager/commit/97193e3b9d225da0166d722a9e74b017d76bfd2c It's apparently  in the library, good news!

tsayles, 2017-01-02:
I also ran into this on Linux Mint 18 with Atom 1.12.7

librepo_fs.js edit from @ybizeul seems to have fixed the problem. 

Steps to reproduce:

1. Open Atom and update all pending spark/Particle* packages;
2. git pull https://github.com/snovalleypa/SVPA-FloodSensor-2016.git dev
   _or clone_
3. Open with ./firmware folder with ATOM
4. Open SVPA-FloodSensor-2016.ino and select Particle --> Compile in the Could from the menu.

**Atom Version**: 1.12.7
**Electron Version**: 1.3.13
**System**: Linux Mint
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.13

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, stat '/home/tsayles/dropbox/projects/openhardware/svpa-floodsensor-2016/firmware/checkupdateota.h'

```
At /home/tsayles/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1233

Error: ENOENT: no such file or directory, stat '/home/tsayles/dropbox/projects/openhardware/svpa-floodsensor-2016/firmware/checkupdateota.h'
    at Error (native)
    at fs.statSync (fs.js:992:18)
    at Object.fs.statSync (ELECTRON_ASAR.js:275:16)
    at normalizeAndSplitPath (/home/tsayles/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1168:16)
    at Object.pathsCommonPrefix (/home/tsayles/.atom/packages/particle-dev/node_modules/particle-library-manager/dist/librepo_fs.js:1220:17)
    at Object.mapCommonPrefix (/home/tsayles/.atom/packages/particle-dev/lib/main.coffee:596:20)
    at /home/tsayles/.atom/packages/particle-dev/lib/main.coffee:630:41
    at Object.loginRequired (/home/tsayles/.atom/packages/particle-dev/lib/main.coffee:300:5)
    at Object.compileCloud (/home/tsayles/.atom/packages/particle-dev/lib/main.coffee:604:65)
    at atom-workspace.<anonymous> (/home/tsayles/.atom/packages/particle-dev/lib/main.coffee:83:55)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/usr/share/atom/resources/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/usr/share/atom/resources/app.asar/src/atom-environment.js:1136:28)
    at EventEmitter.outerCallback (/usr/share/atom/resources/app.asar/src/application-delegate.js:319:25)
    at emitTwo (events.js:106:13)
    at EventEmitter.emit (events.js:191:7)
```

### Commands

```
     -3:47.6.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always)
     -3:47.6.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always)
     -3:46.8.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:46.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:46.7.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:46.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:51.1.0 core:move-left (atom-text-editor.editor.is-focused)
     -1:50.7.0 core:backspace (atom-text-editor.editor.is-focused)
     -1:48.9.0 core:move-down (atom-text-editor.editor.is-focused)
     -1:20.1.0 core:save (atom-text-editor.editor.is-focused)
  2x -0:55.5.0 particle-dev:compile-cloud (atom-text-editor.editor.is-focused)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "particle-dev-libraries"
    ],
    "telemetryConsent": "limited"
  }
[... 84 lines cut ...]
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

suda, 2017-01-03:
@magnum129 @ybizeul @tsayles just relased v0.1.14 which should fix this issue :)

commit 2807237d (2017-01-03): Fix Case Sensitivity issue

Close #172
closed 2017-01-03 by suda