particle-iot-archived/particle-dev #178: Particle account logout/login not showing correct devices

steelydev, 2017-01-24:
Particle Dev 1.7.4

In order to move between Particle Accounts, it is necessary to take the following steps:
Particle->Log Out 'email'. 
Quit Particle Dev
Run Particle Dev  
Particle->Log In to Particle Cloud...

Otherwise Particle Dev shows the devices for the first login.  You can not rename the devices as 'Permission Denied' shows (correctly).

[image]

WiFire_BK is not one of my devices!

suda, 2017-01-24:
Hi @steelydev this should be fixed by https://github.com/spark/particle-dev-profiles/releases/tag/v0.1.14 :) We'll release standalone Particle Dev App soon, but until then you can use Atom and [`particle-dev-complete`](https://atom.io/packages/particle-dev-complete) package.

closed 2017-01-24 by suda

steelydev, 2017-01-25:
@suda thanks.  Can you explain the version numbers?  Is 1.7.4 the Atom version?  Where would I find the Particle version?

suda, 2017-01-26:
Sure! Version of Particle Dev App is a version of Atom it's running. Version of the Particle tools packages can be checked in Settings -> Packages view when searching for `particle-dev`