particle-iot-archived/particle-dev #179: Unable to select a Raspberry Pi as a target; "Cannot read property 'name' of undefined"

legoadk, 2017-01-24:
1. Add a Raspberry Pi to your account as a target.
2. Select it as a target.

**Atom Version**: 1.7.4
**System**: Mac OS X 10.12.2
**Thrown From**: [particle-dev-profiles](https://github.com/spark/particle-dev-profiles) package, v0.1.12

### Stack Trace

Uncaught TypeError: Cannot read property 'name' of undefined

```
At /Applications/Particle%20Dev.app/Contents/Resources/app.asar/node_modules/particle-dev-profiles/src/profiles-tile.js:60

TypeError: Cannot read property 'name' of undefined
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev-profiles/src/profiles-tile.js:38:95
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at ProfileManager.setLocal (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev-profiles/src/profile-manager.js:121:16)
    at ProfileManager.set (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev-profiles/src/profile-manager.js:191:8)
    at SelectCoreView.module.exports.SelectCoreView.confirmed (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/select-core-view.js:76:51)
    at SelectCoreView.module.exports.SelectListView.confirmSelection (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/lib/select-list-view.js:338:21)
    at space-pen-li.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/lib/select-list-view.js:139:19)
    at HTMLOListElement.jQuery.event.dispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLOListElement.elemData.handle (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/jquery/dist/jquery.js:4121:28)
```

### Commands

```
     -2:50.3.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -2:50.3.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -2:50.3.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -2:50.3.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -2:50.1.0 application:add-project-folder (atom-text-editor.editor.is-focused)
     -1:53 particle-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:50.6.0 particle-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:50.6.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:50.6.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:50.3.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
  2x -1:47.8.0 particle-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:34.6.0 particle-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:34.6.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -1:34.6.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
  4x -1:34.4.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
     -0:02.1.0 particle-dev:select-device (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-atom-dark-ui)
```

legoadk, 2017-01-24:
Additionally, "Raspberry Pi" does not show up in the list of platforms that appears when the target platform icon is clicked. I figure that this probably has something to do with it.

suda, 2017-01-24:
Hi Andrew. This bug is fixed in the package itself but requires Particle Dev App (the standalone binary) to be updated. We'll do it in following days but until then you can use Atom and `particle-dev-complete` package.

closed 2017-01-24 by suda