particle-iot-archived/particle-dev #180: [Particle Build] Legacy libraries can't be removed from project

ScruffR, 2017-01-25:


christinestamper, 2017-01-27:
I have had this same problem. I tried Firefox, IE and Chrome. I finally was able to fiddle around in Chrome enough to delete a library. In firefox, sometimes the X didn't show up, same thing in IE. Even when it did, if you point to it to click, a hint text box comes up and blocks the button. In chrome, after some fiddling, I was able to delete. However, I'm hoping for a fix to this soon.

GregSawin, 2017-01-29:
Same issue, can you say what kind of fiddling you did to be able to remove a library?

ScruffR, 2017-01-30:
I was able to do it with FireFox on a highres screen where the fonts were smaller in comparison to the width of the side bar.

GregSawin, 2017-02-07:
I tried shrinking and zooming with no success.  I don't see a more info popup like some have which simply cover the X, it's not showing at all.  I'm tired of waiting so I'm just going to create a new project and copy/paste everything over :(

suda, 2017-02-14:
This has been logged in closed issue tracker for Build

closed 2017-02-14 by suda