particle-iot-archived/particle-dev #183: Uncaught TypeError: Invalid "baudRate" must be a number got: 115200

atypaldos-ias, 2017-02-16:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.2 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.11.6
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package 0.1.19

### Stack Trace

Uncaught TypeError: Invalid "baudRate" must be a number got: 115200

```
At /Users/atypaldos/.atom/packages/particle-dev/node_modules/serialport/lib/serialport.js:101

TypeError: Invalid "baudRate" must be a number got: 115200
    at /packages/particle-dev/node_modules/serialport/lib/serialport.js:101:11)
    at SerialMonitorView.module.exports.SerialMonitorView.connect (/packages/particle-dev/lib/views/serial-monitor-view.coffee:135:17)
    at SerialMonitorView.module.exports.SerialMonitorView.toggleConnect (/packages/particle-dev/lib/views/serial-monitor-view.coffee:123:8)
    at /packages/particle-dev/node_modules/space-pen/lib/space-pen.js:220:36)
    at HTMLButtonElement.dispatch (/packages/particle-dev/node_modules/space-pen/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLButtonElement.elemData.handle (/packages/particle-dev/node_modules/space-pen/node_modules/jquery/dist/jquery.js:4121:28)
```

### Commands

```
     -0:21.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.7.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.7.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:21.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.7.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.7.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:21.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:19.6.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:18.2.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:18.2.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  3x -0:18.1.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Non-Core Packages

```
console-panel 0.2.1 
git-plus 7.3.0 
language-particle 0.3.4 
particle-dev 0.1.19 
particle-dev-cloud-functions 0.1.0 
particle-dev-cloud-variables 0.1.0 
particle-dev-complete 1.0.1 
particle-dev-libraries 0.2.18 
particle-dev-profiles 0.1.14 
tool-bar 1.0.1 
```

keavin7, 2017-02-20:
Also got this. Must be a wrong datatype conversion or something??

zacnelson, 2017-02-21:
+1 also encountered this issue. Here's my Stack Trace:

[Enter steps to reproduce below:]

1. Select baud rate after opening USB monitor
2. Click "connect"

**Atom Version**: 1.7.4
**System**: Microsoft Windows 7 Professional 
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.18

### Stack Trace

Uncaught TypeError: Invalid "baudRate" must be a number got: 9600

```
At C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\particle-dev\node_modules\serialport\lib\serialport.js:101

TypeError: Invalid "baudRate" must be a number got: 9600
    at new SerialPort (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\particle-dev\node_modules\serialport\lib\serialport.js:101:11)
    at SerialMonitorView.module.exports.SerialMonitorView.connect (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\particle-dev\lib\views\serial-monitor-view.js:229:19)
    at SerialMonitorView.module.exports.SerialMonitorView.toggleConnect (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\particle-dev\lib\views\serial-monitor-view.js:213:21)
    at HTMLButtonElement.<anonymous> (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:220:36)
    at HTMLButtonElement.jQuery.event.dispatch (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\jquery\dist\jquery.js:4435:9)
    at HTMLButtonElement.elemData.handle (C:\Program Files (x86)\ParticleDev\resources\app.asar\node_modules\jquery\dist\jquery.js:4121:28)
```

### Commands

```
     -0:24.5.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
  2x -0:24.5.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:24.5.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:24.5.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:24.5.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
  2x -0:24.1.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:19.6.0 particle-dev:show-serial-monitor (button.btn.pull-right)
     -0:17.1.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.2.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.2.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.2.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
  2x -0:16.2.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.1.0 particle-dev:update-login-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.1.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:16.1.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
  2x -0:15.8.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```

alanmalk, 2017-02-27:
Same problem...

bioagbob, 2017-03-09:
Had same error tonight for the first time while trying to use baud other than 9600.  MacOS 10.11.6, P Dev 0.1.20.  
Workaround for me
_1. set baud to 9600  
2. exit app  
3. launch app
4. open serial port at 9600_  
Key is that I can only use 9600.  I reproduced the error a couple times.  Any time I tried a rate other than 9600, it failed with this 'Invalid "BaudRate"' error.  Subsequently trying 9600 in the same session also failed.  But after restarting Particle Dev, it worked, but only if I set to 9600 BEFORE quitting P Dev and re-launching.

commit 5990475f (2017-07-18): Fix #183
closed 2017-07-18 by suda