particle-iot-archived/particle-dev #191: Atom is out of date

GordonRudman, 2017-02-28:
I constantly get errors when trying to compile, saying that Atom is out of date.
The error was thrown from the particle-dev package. Atom is out of date: 1.8.0 installed; 1.14.4 latest. Upgrading to the latest version may fix this issue.

Uncaught TypeError: Cannot read property 'replace' of undefined

C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:59
Hide Stack Trace
TypeError: Cannot read property 'replace' of undefined
    at CompileErrorsView.module.exports.CompileErrorsView.fixInoFile (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:59:22)
    at Object.content (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:89:21)
    at Builder.tag (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:281:19)
    at Function.View.(anonymous function) [as li] (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:83:49)
    at Function.<anonymous> (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:80:21)
    at Function.View.buildHtml (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:123:10)
    at Function.View.render (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:129:19)
    at exports.$$ (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\node_modules\space-pen\lib\space-pen.js:627:24)
    at CompileErrorsView.module.exports.CompileErrorsView.viewForItem (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:79:14)
    at CompileErrorsView.module.exports.SelectListView.populateList (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\lib\select-list-view.js:206:29)
    at CompileErrorsView.module.exports.SelectListView.setItems (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\atom-space-pen-views\lib\select-list-view.js:167:12)
    at CompileErrorsView.module.exports.CompileErrorsView.show (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:69:14)
    at CompileErrorsView.show (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\views\compile-errors-view.js:3:61)
    at Object.module.exports.showCompileErrors (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1132:37)
    at atom-workspace.<anonymous> (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:131:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:160:19)
    at C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1019:29
    at Object.module.exports.minBuildTargetRequired (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:498:14)
    at C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:941:24
    at Object.module.exports.loginRequired (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:450:14)
    at Object.module.exports.compileCloud (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:939:19)
    at atom-workspace.<anonymous> (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:160:19)
    at executeCallback (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (C:\Users\grudman\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:82:9)

GonzRon, 2017-03-03:
I am also getting this error, tried uninstalling / reinstalling manually, no dice.

here's my stack trace:

`TypeError: path must be a string
    at TypeError (native)
    at Object.fs.openSync (fs.js:584:18)
    at Object.module.(anonymous function) [as openSync] (ATOM_SHELL_ASAR.js:159:20)
    at Object.fs.readFileSync (fs.js:431:33)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:427:29)
    at C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1030:40
    at Object.module.exports.minBuildTargetRequired (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:498:14)
    at C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:941:24
    at Object.module.exports.loginRequired (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:450:14)
    at Object.module.exports.compileCloud (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:939:19)
    at atom-workspace.<anonymous> (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:160:19)
    at executeCallback (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (C:\Users\Lcstyle\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:82:9)`

suda, 2017-03-03:
@GordonRudman It seems that this is a bug in the compile code. Could you provide the the code that causes this issue?

Tico42, 2017-03-06:
im trying to work with this sensor but im getting the same problem
You'll need all the liberies(<FreqCount.h> and <MD_TCS230.h>) i suggest download them from this site is where i got them in the first place.
http://www.mediafire.com/file/pn75hpb13x1314v/MD_TCS230_v1.0_20130222.zip
Im also using a photon.
------------------------------------------------------------------------------------
// TCS230 sensor calibration and color readings
//
// Input and output using the Serial console.
//
#include <MD_TCS230.h>
#include <FreqCount.h>

#define	BLACK_CAL	0
#define	WHITE_CAL	1
#define	READ_VAL	2

// Pin definitions
#define  S2_OUT  3
#define  S3_OUT  4
#define  OE_OUT   5    // LOW = ENABLED 

MD_TCS230	CS(S2_OUT, S3_OUT, OE_OUT);

void setup()
{
  Serial.begin(57600);
  Serial.print(F("\n[TCS230 Calibrator Example]"));

  CS.begin();
}

char getChar()
// blocking wait for an input character from the input stream
{
	while (Serial.available() == 0)
		;
	return(toupper(Serial.read()));
}

void clearInput()
// clear all characters from the serial input
{
	while (Serial.read() != -1)
		;
}

uint8_t fsmReadValue(uint8_t state, uint8_t valType, uint8_t maxReads)
// Finite State Machine for reading a value from the sensor
// Current FSM state is passed in and returned
// Type of value being read is passed in
{
	static	uint8_t	selChannel;
	static	uint8_t	readCount;
	static	sensorData	sd;

	switch(state)
	{
	case 0:	// Prompt for the user to start
		Serial.print(F("\n\nReading value for "));
		switch(valType)
		{
		case BLACK_CAL:	Serial.print(F("BLACK calibration"));	break;
		case WHITE_CAL:	Serial.print(F("WHITE calibration"));	break;
		case READ_VAL:	Serial.print(F("DATA"));				break;
		default:		Serial.print(F("??"));					break;
		}

		Serial.print(F("\nPress any key to start ..."));
		state++;
		break;

	case 1:	// Wait for user input
		getChar();
		clearInput();
		state++;
		break;

	case 2:	// start the reading process
		CS.read();
		state++;
		break;

	case 3:	// wait for a read to complete
		if (CS.available())
		{
			sensorData	sd;
			colorData	rgb;

			switch(valType)
			{
			case BLACK_CAL:
				CS.getRaw(&sd);
				CS.setDarkCal(&sd);
				break;

			case WHITE_CAL:
				CS.getRaw(&sd);
				CS.setWhiteCal(&sd);
				break;

			case READ_VAL:
				CS.getRGB(&rgb);
				Serial.print(F("\nRGB is ["));
				Serial.print(rgb.value[TCS230_RGB_R]);
				Serial.print(F(","));
				Serial.print(rgb.value[TCS230_RGB_G]);
				Serial.print(F(","));
				Serial.print(rgb.value[TCS230_RGB_B]);
				Serial.print(F("]"));
				break;
			}
			state++;
		}
		break;

	default:	// reset fsm
		state = 0;
		break;
	}

	return(state);
}

void loop()
{
  static uint8_t	runState = 0;
  static uint8_t	readState = 0;

  switch(runState)
  {
  case 0:	// calibrate black
	  readState = fsmReadValue(readState, BLACK_CAL, 2);
	  if (readState == 0) runState++;
	  break;

  case 1:	// calibrate white
	  readState = fsmReadValue(readState, WHITE_CAL, 2);
	  if (readState == 0) runState++;
	  break;

  case 2:	// read color
	  readState = fsmReadValue(readState, READ_VAL, 1);
	  break;

  default:
	  runState = 0;	// start again if we get here as something is wrong
  }
}

GonzRon, 2017-03-14:
these are not code related errors, I have code that was compiling perfectly and now won't compile with particle dev 1.8 nor 1.7, I believe these are atom / electron related dependency issues that need to be sorted out by the particle-dev developers.

SammyHerring, 2017-03-22:
Hi, I am also having these issues with compiling. As far as I am aware, there are no issues with the code. In the meantime, while a fix is produced, can anyone recommend a workaround? Thank you,

Error Message:
`TypeError: Cannot read property 'replace' of undefined
    at CompileErrorsView.module.exports.CompileErrorsView.fixInoFile (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:59:22)
    at Object.content (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:89:21)
    at Builder.tag (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/node_modules/space-pen/lib/space-pen.js:281:19)
    at Function.View.(anonymous function) [as li] (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/node_modules/space-pen/lib/space-pen.js:83:49)
    at Function.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:80:21)
    at Function.View.buildHtml (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/node_modules/space-pen/lib/space-pen.js:123:10)
    at Function.View.render (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/node_modules/space-pen/lib/space-pen.js:129:19)
    at exports.$$ (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/node_modules/space-pen/lib/space-pen.js:627:24)
    at CompileErrorsView.module.exports.CompileErrorsView.viewForItem (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:79:14)
    at CompileErrorsView.module.exports.SelectListView.populateList (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/lib/select-list-view.js:206:29)
    at CompileErrorsView.module.exports.SelectListView.setItems (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/atom-space-pen-views/lib/select-list-view.js:167:12)
    at CompileErrorsView.module.exports.CompileErrorsView.show (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:69:14)
    at CompileErrorsView.show (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/views/compile-errors-view.js:3:61)
    at Object.module.exports.showCompileErrors (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:1132:37)
    at atom-workspace.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:131:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:160:19)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:1019:29
    at Object.module.exports.minBuildTargetRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:498:14)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:941:24
    at Object.module.exports.loginRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:450:14)
    at Object.module.exports.compileCloud (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:939:19)
    at atom-workspace.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/atom-environment.js:1100:28)
    at EventEmitter.outerCallback (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/application-delegate.js:327:25)
    at emitTwo (events.js:87:13)
    at EventEmitter.emit (events.js:172:7)`

jventerpriese, 2017-03-22:
I too started getting this message a few moments ago right after i copied my source.cpp to source_copy.cpp in the project directory, but did not include or reference it in the project.

as soon as i moved the unused copied file out of the project directory the error went away

asiraky, 2017-03-26:
I was able to compile by deleting the electron_0.6.0_firmware_1234567788.bin file that was in the directory open in Particle Dev. Hope that helps someone.

kvwilson, 2017-06-10:
I just got this today as well... I think it may have been due to an extra file I left in the project dir... I brought back a file history backup of a .h file and made a copy of the original in case I had regrets.  As soon as I removed the copy of that file, it worked.  sounds like black magic but I'll take it.

Eco-Ants, 2017-07-19:
Yes, if you remove other files from the folder, this error goes away :)
It would be nice if Particle FIXED this bug, as I'm sure it's frustrated a lot of people and wasted a lot of time !!!

suda, 2017-07-20:
This seems to be fixed in [v0.1.22](https://github.com/spark/particle-dev/releases/tag/v0.1.22). If you're using standalone Particle Dev App, we're going to release an update soon that will bring this fix to you. Until then [installing Particle Dev in Atom](https://github.com/spark/particle-dev#github-atom) could help.

closed 2017-07-20 by suda