particle-iot-archived/particle-dev #195: copied libraries contain a .h file that prevents compiling

cmsunu28, 2017-03-09:
Copying a library in dev creates a structure with a second .h file nested in a folder. This throws an error as a default and prevents the library from compiling until it is removed.

Copy library
[image]

Won't compile immediately after, no edits to library or anything
[image]

Here's the problem
[image]

Deleted that .h and moved the other library files inside
[image]

m-mcgowan, 2017-03-10:
Thanks for the detailed description and screen grabs.

The nested includes are "adapters" that support legacy applications using the `#include "LibraryName/LibraryName.h"` syntax. They are meant to be there, and shouldn't cause an issue themselves if the rest of the library is also present. 

The actual header file from the library should exist in the `src` folder of the library, but from the screenshot, it looks like that it's missing.

m-mcgowan, 2017-03-10:
I just tried this and was unable to replicate the directory structure above. This is what I get after copying the library:

[image]

I'm not sure how to explain the difference:
- are we using the same version of particle-dev-libraries? I'm on 0.2.18
- Is there any chance you may have dragged those files from `src` into the Adafruit_MLX90614 subfolder? In your screenshot, there are two files in the subfolder, where there should only be one.

ScruffR, 2017-06-08:
The parallel issue is already confirmed, so I guess this should be too
https://github.com/spark/particle-dev-libraries/issues/26