particle-iot-archived/particle-dev #197: Locally installed core Atom package particle-dev is out of date

prlarson, 2017-03-11:
I'm getting this error when I try to compile.  I have removed all related files, prefs, etc., and reinstalled.  What am I missing?

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: Mac OS X 10.12.3
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.20

### Stack Trace

Uncaught TypeError: path must be a string

```
At fs.js:584

TypeError: path must be a string
    at TypeError (native)
    at Object.fs.openSync (fs.js:584:18)
    at Object.module.(anonymous function) [as openSync] (ATOM_SHELL_ASAR.js:159:20)
    at Object.fs.readFileSync (fs.js:431:33)
    at Object.fs.readFileSync (ATOM_SHELL_ASAR.js:427:29)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:1030:40
    at Object.module.exports.minBuildTargetRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:498:14)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:941:24
    at Object.module.exports.loginRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:450:14)
    at Object.module.exports.compileCloud (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:939:19)
    at atom-workspace.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:160:19)
    at executeCallback (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/tool-bar/lib/tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/tool-bar/lib/tool-bar-button-view.js:82:9)
```

### Commands

```
  2x -7:32.1.0 core:copy (atom-notification.fatal.icon.icon-bug.native-key-bindings.has-detail.has-close.has-stack)
     -2:51.3.0 settings-view:uninstall-packages (atom-notification.fatal.icon.icon-bug.native-key-bindings.has-detail.has-close.has-stack)
     -0:00.7.0 particle-dev:compile-cloud (atom-text-editor.editor)
     -0:00.7.0 window:save-all (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
     -0:00.7.0 particle-dev:update-compile-status (atom-workspace.workspace.scrollbars-visible-always.theme-monokai.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
No installed packages

# Dev
No dev packages
```

GonzRon, 2017-03-14:
This is a blocking issue, I cannot develop for particle photon on desktop env. I even tried 1.7.4 and still get build errors. I think these problems are related to atom / other particle dev dependency issues.

suda, 2017-03-14:
That is possible. We'll look into it asap

trunet, 2017-03-28:
Are there any workaround for this issue?

trunet, 2017-03-29:
Ok, I found a workaround. If you have more than one project on your tree, this error happens. Just let one project open, and it should work just fine.

prlarson, 2017-03-29:
unfortunately, only have one project open did not work for me

suda, 2017-04-04:
@prlarson could you upload a screenshot of your the project tree?

prlarson, 2017-04-06:
screen shot:
[image]

suda, 2017-04-06:
This should be fixed in latest version of [Particle Dev package](https://github.com/spark/particle-dev/releases/tag/v0.1.21). We're going to update the standalone app this week but until then you can try installing the package in Atom.

The reason of your error is that main file doesn't have an extension. Once you add `.ino` or `.cpp` it should work :)

closed 2017-04-06 by suda

zaprime, 2017-07-06:
Did you fix the bug? The last release on the Git repository was 21 February?

suda, 2017-07-06:
[v0.1.21](https://github.com/spark/particle-dev/releases/tag/v0.1.21) was released on the 5th of April with a fix for this.