particle-iot-archived/particle-dev #198: atom install particle-dev doesn't work

GonzRon, 2017-03-14:
I am trying to workaround compile problems in the standard winx64 distribution archive that is provided from the particle dev official site by installing latest version of atom and then using the package manager to install particle-dev or particle-dev-complete, after struggling for many hours, I realized that there are many depends (including git for windows, python, etc.)  After fixing all of those issues, I finally got the error log down to a point where I can no longer figure out what to do next.

the atom log errors with the following:
```
D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\bin> .\apm install particle-dev
Installing particle-dev to C:\Users\Lcstyle\.atom\packages failed

> serialport@4.0.1 install C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote\node_modules\particle-dev\node_modules\serialport
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB3428: Could not load the Visual C++ component "VCBuild.exe". To fix this, 1) install the .NET Framework 2.0 SDK, 2) install Microsoft Visual Studio 2005 or 3) add the location of the component to the system path if it is installed elsewhere.  [C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote\node_modules\particle-dev\node_modules\serialport\build\binding.sln]
Failed to execute 'D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\bin\node.exe D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote\node_modules\particle-dev\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote\node_modules\particle-dev\node_modules\serialport\build\Release' (1)
C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote
`-- (empty)

npm WARN deprecated spark@0.5.10: Use particle-api-js instead
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN excluding symbolic link test\server\static\ie8-polyfill.js -> ../../../ie8-polyfill.js
(node) warning: possible EventEmitter memory leak detected. 11 error listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at TLSSocket.addListener (events.js:239:17)
    at TLSSocket.Readable.on (_stream_readable.js:673:33)
    at Request.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\npm-registry-client\lib\request.js:153:7)
    at emitOne (events.js:77:13)
    at Request.emit (events.js:169:7)
    at ClientRequest.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\request\request.js:837:10)
    at emitOne (events.js:82:20)
    at ClientRequest.emit (events.js:169:7)
    at tickOnSocket (_http_client.js:502:7)
    at onSocketNT (_http_client.js:514:5)
(node) warning: possible EventEmitter memory leak detected. 11 error listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at TLSSocket.addListener (events.js:239:17)
    at TLSSocket.Readable.on (_stream_readable.js:673:33)
    at Request.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\npm-registry-client\lib\request.js:153:7)
    at emitOne (events.js:77:13)
    at Request.emit (events.js:169:7)
    at ClientRequest.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\request\request.js:837:10)
    at emitOne (events.js:82:20)
    at ClientRequest.emit (events.js:169:7)
    at tickOnSocket (_http_client.js:502:7)
    at onSocketNT (_http_client.js:514:5)
(node) warning: possible EventEmitter memory leak detected. 11 error listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at TLSSocket.addListener (events.js:239:17)
    at TLSSocket.Readable.on (_stream_readable.js:673:33)
    at Request.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\npm-registry-client\lib\request.js:153:7)
    at emitOne (events.js:77:13)
    at Request.emit (events.js:169:7)
    at ClientRequest.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\request\request.js:837:10)
    at emitOne (events.js:82:20)
    at ClientRequest.emit (events.js:169:7)
    at tickOnSocket (_http_client.js:502:7)
    at onSocketNT (_http_client.js:514:5)
(node) warning: possible EventEmitter memory leak detected. 11 error listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at TLSSocket.addListener (events.js:239:17)
    at TLSSocket.Readable.on (_stream_readable.js:673:33)
    at Request.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\npm-registry-client\lib\request.js:153:7)
    at emitOne (events.js:77:13)
    at Request.emit (events.js:169:7)
    at ClientRequest.<anonymous> (D:\Downloads\atom-x64-windows\Atom x64\resources\app\apm\node_modules\request\request.js:837:10)
    at emitOne (events.js:82:20)
    at ClientRequest.emit (events.js:169:7)
    at tickOnSocket (_http_client.js:502:7)
    at onSocketNT (_http_client.js:514:5)
(node) warning: possible EventEmitter memory leak detected. 11 error listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at TLSSocket.addListener (events.js:239:17)
[... 151 lines cut ...]
npm ERR! argv "D:\\Downloads\\atom-x64-windows\\Atom x64\\resources\\app\\apm\\bin\\node.exe" "D:\\Downloads\\atom-x64-windows\\Atom x64\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\Lcstyle\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\Lcstyle\\.atom\\.apmrc" "install" "C:\\Users\\Lcstyle\\AppData\\Local\\Temp\\d-117214-15408-1xucdi9\\package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=x64" "--global-style"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Lcstyle\AppData\Local\Temp\apm-install-dir-117214-15408-lz1ote\npm-debug.log
npm ERR! code 1`

```

GonzRon, 2017-03-14:
notice the following error:

node-pre-gyp ERR! Tried to download(404): 
https://github.com/suda/node-serialport/releases/download/4.0.1/serialport-v4.0.1-electron-v1.3-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp)

serialport v4.0.1 electron v1.3 does not exist.

GonzRon, 2017-03-14:
are there any guides to getting particle development going in eclipse or an alternative IDE?

suda, 2017-03-15:
Have you tried installing standalone [Particle Dev App](https://www.particle.io/dev)? You could also try to integrate Eclipse/Netbeans or any IDE using our [CLI](particle.io/cli). There are some tutorials on our community like this one: [Local development and gdb debugging with NetBeans](https://community.particle.io/t/local-development-and-gdb-debugging-with-netbeans-a-step-by-step-guide/7829)

IgorGanapolsky, 2017-07-14:
I am also getting this error in standalone Particle Dev App on Mac:
> [Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: Mac OS X 10.12.5
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package, v0.1.20

### Stack Trace

Uncaught TypeError: Path must be a string. Received undefined

```
At path.js:8

TypeError: Path must be a string. Received undefined
    at assertPath (path.js:8:11)
    at Object.posix.resolve (path.js:426:5)
    at Object.module.exports.processDirIncludes (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:612:22)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:974:29
    at Object.module.exports.minBuildTargetRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:498:14)
    at /Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:941:24
    at Object.module.exports.loginRequired (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:450:14)
    at Object.module.exports.compileCloud (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:939:19)
    at atom-workspace.<anonymous> (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/particle-dev/lib/main.js:146:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Particle Dev.app/Contents/Resources/app.asar/src/command-registry.js:160:19)
    at executeCallback (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/tool-bar/lib/tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (/Applications/Particle Dev.app/Contents/Resources/app.asar/node_modules/tool-bar/lib/tool-bar-button-view.js:82:9)
```

### Commands

```
     -4:21.8.0 command-palette:toggle (atom-text-editor.editor.is-focused)
     -4:18.8.0 core:move-down (atom-text-editor.editor.mini.is-focused)
     -4:15.8.0 core:move-up (atom-text-editor.editor.mini.is-focused)
 14x -4:14 core:backspace (atom-text-editor.editor.mini.is-focused)
     -4:04.4.0 core:confirm (atom-text-editor.editor.mini.is-focused)
     -4:04.4.0 settings-view:install-packages-and-themes (atom-text-editor.editor)
     -4:01.1.0 core:confirm (atom-text-editor.editor.mini.is-focused)
     -3:07.3.0 core:copy (div.alert.alert-danger.alert-dismissable.native-key-bindings)
     -0:55.3.0 particle-dev:select-device (atom-text-editor.editor.mini)
     -0:53.7.0 editor:consolidate-selections (atom-text-editor.editor.mini.is-focused)
     -0:53.7.0 core:cancel (atom-text-editor.editor.mini.is-focused)
     -0:51 core:paste (atom-text-editor.editor.is-focused)
     -0:50.3.0 core:undo (atom-text-editor.editor.is-focused)
     -0:46.6.0 particle-dev:compile-cloud (atom-text-editor.editor)
     -0:35 core:copy (atom-notification.fatal.icon.icon-bug.native-key-bindings.has-detail.has-close.has-stack)
  2x -0:09.8.0 particle-dev:compile-cloud (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-monokai.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "themes": [
      "one-dark-ui",
      "monokai"
    ]
  }
}
```

### Installed Packages

```coffee
# User
build-arduino, v0.1.1
build-processing, v0.1.2
language-arduino, v0.4.3
language-kotlin, v0.5.0
linter, v1.11.18
linter-kotlin, v0.5.0
particle-dev-complete, v1.0.2
particle-offline-compiler, v0.1.3

# Dev
No dev packages
```

suda, 2017-07-18:
@IgorGanapolsky doesn't seem to be connected to package installation problems mentioned here. Could you open a separate issue for this?

suda, 2017-07-18:
@GonzRon we updated the instructions for the installation: https://github.com/spark/particle-dev#github-atom (they should also work with the x64 Windows). Are you able to make it work using them?