particle-iot-archived/particle-dev #199: atom particle-dev does not install on Windows

fhiggins, 2017-03-15:
I followed the instructions in the readme - one link now takes you to Visual Studio Community 2017.

After loading these requirements, when loading particle-dev from atom, the error is that it cannot find cl.exe. The file exists, but obviously the path variable is not set correctly. However, Visual Studio runs fine. Do you have recommendations as to how to best set the path - or is there something in starting atom that a user can change in that startup environment?

Your current instructions do not result in an error free installation.

suda, 2017-03-15:
I'm sorry to hear that. Do you get the same errors as #198? I could suggest installing the standalone [Particle Dev App](https://www.particle.io/dev) which doesn't need VS or any other dependencies.

fhiggins, 2017-03-15:
Thanks for your reply - no I don't get the same errors.

As I pointed out, the first error was that the tool couldn't find CL.exe. I found the file and added it to my PATH variable. Retried the package install. Now it can't find stdio.h, and others - so the include path is not clean. However, VS 2017 does not have these files, so I can't even define an INCLUDE variable.
I also tried to look for VS Community 2013 as it probably has all of these things in place, but cannot find it.

Bottom line, if you're going to depend on MS VS Community tool, it appears that it changes a fair amount from version to version - and since one cannot load previous versions - this may be an ongoing challenge for you.

I will look at the standalone app. Thanks for the input.

closed 2017-03-15 by fhiggins