particle-iot-archived/particle-dev #200: Uncaught TypeError: Cannot read property 'length' of null

zfields, 2017-03-19:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.15.0 x64
**Electron**: 1.3.13
**OS**: Ubuntu 16.04.2
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package 0.1.20

### Stack Trace

Uncaught TypeError: Cannot read property 'length' of null

```
At /home/zak/.atom/packages/particle-dev/lib/main.coffee:856

TypeError: Cannot read property 'length' of null
    at /packages/particle-dev/lib/main.coffee:856:13
    at /packages/particle-dev/lib/main.coffee:341:7
    at Object.loginRequired (/packages/particle-dev/lib/main.coffee:322:5)
    at Object.deviceRequired (/packages/particle-dev/lib/main.coffee:326:6)
    at Object.flashCloud (/packages/particle-dev/lib/main.coffee:829:35)
    at /packages/particle-dev/lib/main.coffee:91:53)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/app.asar/src/atom-environment.js:1162:28)
    at EventEmitter.outerCallback (/app.asar/src/application-delegate.js:312:25)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
     -4:07.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:45.3.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:45.3.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:27.1.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -3:27.1.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -2:00.7.0 particle-dev:select-device (input.hidden-input)
     -1:56.7.0 particle-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:56.7.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:56.7.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  3x -1:56.2.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:42.5.0 particle-dev:select-device (input.hidden-input)
     -0:40 particle-dev:update-core-status (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:40 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:40 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:39.6.0 particle-dev:core-status-updated (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:11.7.0 particle-dev:flash-cloud (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Non-Core Packages

```
console-panel 0.2.1 
particle-dev 0.1.20 
particle-dev-cloud-functions 0.1.0 
particle-dev-cloud-variables 0.1.0 
particle-dev-complete 1.0.1 
particle-dev-libraries 0.2.18 
particle-dev-profiles 0.1.15 
tool-bar 1.1.0 
```

suda, 2017-03-20:
Thank for filing this Zachary! Could you tell us more about how we could reproduce it?

zfields, 2017-03-21:
I have no experience with Atom. I installed it on Linux, added the particle package, tried to run it, got on error, then it gave me a button to report the issue - so I did.

suda, 2017-03-22:
Could you take a screenshot of the Atom window when the error occurs?

closed 2017-04-05 by suda

adsurg, 2017-05-06:
I've just seen this on Windows.  It looks like it doesn't like compiling an unsaved file.  Stack trace follows.

C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1175
Hide Stack Trace
TypeError: Cannot read property 'length' of null
    at C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1175:20
    at C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:475:18
    at Object.module.exports.loginRequired (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:450:14)
    at Object.module.exports.deviceRequired (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:453:19)
    at Object.module.exports.flashCloud (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:1138:19)
    at atom-workspace.<anonymous> (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\particle-dev\lib\main.js:151:24)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\src\command-registry.js:160:19)
    at executeCallback (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:123:19)
    at ToolBarButtonView._onClick (C:\Users\Adam\AppData\Local\particledev\app-1.8.0\resources\app.asar\node_modules\tool-bar\lib\tool-bar-button-view.js:82:9)

commit ebfa528b (2017-04-05): Fix #200