particle-iot-archived/particle-dev #201: Deprecated selector in `particle-dev\styles\particle-dev.less`

heepdog, 2017-03-21:
In `particle-dev\styles\particle-dev.less`: 

Starting from Atom v1.13.0, the contents of `atom-text-editor` elements are no longer encapsulated within a shadow DOM boundary. This means you should stop using `:host` and `::shadow` pseudo-selectors, and prepend all your syntax selectors with `syntax--`. To prevent breakage with existing style sheets, Atom will automatically upgrade the following selectors:

* `atom-text-editor::shadow .password-lines .line span.text:before` => `atom-text-editor.editor .password-lines .line span.syntax--text:before`

* `atom-text-editor::shadow .password-lines .line span.text` => `atom-text-editor.editor .password-lines .line span.syntax--text`

Automatic translation of selectors will be removed in a few release cycles to minimize startup time. Please, make sure to upgrade the above selectors as soon as possible.

heepdog, 2017-03-21:
Quick little issue to fix the pesky "Depreciated" link.

suda, 2017-03-23:
Closing this as a duplicate of #176 but thank you for adding more info!

closed 2017-03-23 by suda