particle-iot-archived/particle-dev #203: Compiler error messages not showing in desktop IDE

4ULjPwGBOC33N, 2017-03-29:
Compile the test program below (note typo for "void" in the desktop IDE (atom 1.15.0, particle-dev 0.1.20, particle-dev-cloud-functions 0.1.0)

Select "Compile in the cloud". The IDE suppresses the error message and instead returns "Success" in the status bar.

Compile in the web IDE and the compile errors are correctly reported

```
int led2=D7;
Void setup(){
  pinMode(led2,OUTPUT);
}
void loop(){
  digitalWrite(led2,HIGH);
  delay(1000);
  digitalWrite(led2,LOW);
  delay(2000); 
}
```

suda, 2017-03-31:
Hello, I'm unable to reproduce the issue:
[image]

What's the directory structure of your project?

4ULjPwGBOC33N, 2017-03-31:
[image]

4ULjPwGBOC33N, 2017-03-31:
Found the problem. Compile is looking at Test.ino (which has stubs but no link to Test.cpp) , code is in test.cpp.  Is there a compile switch so that I can use regular C++ file structure ?

suda, 2017-04-03:
If you have both `.ino` file and `.cpp` file in your `src` directory, both will be compile. I would suggest moving the files which aren't supposed to be part of the main app itself to a different directory.

closed 2017-07-06 by suda