particle-iot-archived/particle-dev #210: Uncaught TypeError: _this.notificationEl.querySelector is not a function

amoux, 2017-05-28:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.17.2 x64
**Electron**: 1.3.15
**OS**: Ubuntu 16.04.2
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package 0.1.21

### Stack Trace

Uncaught TypeError: _this.notificationEl.querySelector is not a function

```
At /home/leo/.atom/packages/particle-dev/node_modules/atom-package-deps/lib/view.js:22

TypeError: _this.notificationEl.querySelector is not a function
    at /packages/particle-dev/node_modules/atom-package-deps/lib/view.js:22:58
```

### Commands

```
     -1:43.7.0 core:confirm (input.hidden-input)
  3x -1:25.7.0 core:backspace (input.hidden-input)
     -1:23.4.0 core:confirm (input.hidden-input)
 19x -1:08.6.0 core:backspace (input.hidden-input)
     -1:04.1.0 core:confirm (input.hidden-input)
     -0:09.3.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:09.3.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:09.3.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Non-Core Packages

```
console-panel 0.2.1 
particle-dev 0.1.21 
particle-dev-complete 1.0.1 
tool-bar 1.1.0 
```

bu5hm4nn, 2017-06-12:
I get this error when I try to install the `linter-gcc` package for Atom.

suda, 2017-07-06:
This might be caused by an old version of `atom-package-deps` package. We'll update it and release the packages using it.

suda, 2017-07-12:
We've updated this dependency. @bu5hm4nn does this issue persist?

bu5hm4nn, 2017-07-12:
@suda Unfortunately it does persist. I can install `linter-gcc` but when I reopen Atom the error pops up again.

suda, 2017-07-13:
Could you paste the contents (including the stacktrace) of the error here? The sources mentioned in the original issue shouldn't exist anymore.

bu5hm4nn, 2017-07-14:
[Enter steps to reproduce:]

1. Install particle-dev
2. Install linter-gcc

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.5
**Thrown From**: [particle-dev](https://github.com/spark/particle-dev) package 0.1.23

### Stack Trace

Uncaught TypeError: _this.notificationEl.querySelector is not a function

```
At /Users/Nethanja/.atom/packages/particle-dev/node_modules/atom-package-deps/lib/view.js:36

TypeError: _this.notificationEl.querySelector is not a function
    at /packages/particle-dev/node_modules/atom-package-deps/lib/view.js:36:62
```

### Commands

```
     -0:16.9.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
     -0:16.9.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
     -0:16.8.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
     -0:16.8.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
     -0:16.8.0 particle-dev:update-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
     -0:16.8.0 particle-dev:append-menu (atom-workspace.workspace.scrollbars-visible-always.theme-one-light-syntax.theme-one-light-ui)
```

### Non-Core Packages

```
atom-beautify 0.30.3 
console-panel 0.2.1 
language-particle 0.3.5 
linter-gcc 0.7.1 
particle-dev 0.1.23 
particle-dev-cloud-functions 0.1.1 
particle-dev-cloud-variables 0.1.1 
particle-dev-complete 1.0.2 
particle-dev-libraries 0.2.19 
particle-dev-profiles 0.2.1 
particle-dev-release-notes 0.53.2 
tool-bar 1.1.0 
```

suda, 2017-07-14:
Hmm I did a fresh install of `particle-dev` packages, then installed `linter-gcc` and got `Uncaught TypeError: _this.notificationEl.querySelector is not a function` but originating from `linter-gcc`. It seems to be a [known issue](https://github.com/hebaishi/linter-gcc/issues/126) and I [created a PR that should fix it](https://github.com/hebaishi/linter-gcc/pull/137). Until then you can first install `linter` package in Atom/Particle Dev App to make this message go away.

closed 2017-07-14 by suda